coolchip/node-red-contrib-smartmeter #1: Uncaught Exception

twimon, 2017-08-31:
31 Aug 09:21:09 - [red] Uncaught Exception:
31 Aug 09:21:09 - Error: Error while parsing SML message: Error: Unknown TL-Field 0x10 for OctetString [Offset: 81]!: Message: 1b1b1b1b01010101760700090a29729c620062007263010176010107000908ba26340b06454d480104c56bc655010163baa700760700090a29729d620062007263070177010b06454d4801048104010010824a4800091b0109760a6b810b0172291b1b01001b1b1b01010101760700090a2972ae620062c60772085200010745750a01011b1b1b1b01010101
    at SmlProtocol.handleMessage (/usr/lib/node_modules/node-red-contrib-smartmeter/node_modules/smartmeter-obis/lib/protocols/SmlProtocol.js:50:15)
    at SerialPort.<anonymous> (/usr/lib/node_modules/node-red-contrib-smartmeter/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:63:41)
    at emitOne (events.js:96:13)
    at SerialPort.emit (events.js:188:7)
    at SerialPort.module.exports.raw (/usr/lib/node_modules/node-red-contrib-smartmeter/node_modules/smartmeter-obis/node_modules/serialport/lib/parsers.js:7:13)
    at SerialPort._emitData (/usr/lib/node_modules/node-red-contrib-smartmeter/node_modules/smartmeter-obis/node_modules/serialport/lib/serialport.js:313:18)
    at SerialPort.<anonymous> (/usr/lib/node_modules/node-red-contrib-smartmeter/node_modules/smartmeter-obis/node_modules/serialport/lib/serialport.js:293:14)
    at SerialPort.<anonymous> (/usr/lib/node_modules/node-red-contrib-smartmeter/node_modules/smartmeter-obis/node_modules/serialport/lib/serialport.js:306:7)
    at FSReqWrap.wrapper [as oncomplete] (fs.js:681:17)

coolchip, 2017-08-31:
Thanks for this issue. Which meter do you use?

twimon, 2017-08-31:
I have a ED300L from EMH. As a workaround I commented the line 50 with the error out and changed the if statement to validate smlFile
if (crcCheck && smlFile) {

coolchip, 2017-09-01:
Thank you for the correction suggestion. I'm on vacation for a week. Then I will take a closer look at the error.

coolchip, 2017-09-10:
@Apollon77
Could you help me with this issue? It looks like there's a problem with your module.

Apollon77, 2017-09-11:
@twimon Are you sure the message is complete and there are no data receive errors? This is an error in the message itself, so it do not conform to the specs. Most of such errors were data transfer problems.
Is the error the same and on the same position when you try multiple times?

Please check this and report back, if it is really the same always then I will have a deeper look into it.

twimon, 2017-09-11:
I am not able to reproduce the issue after reinstall of the raspberry pi. Probably there was a issue with the installation.

closed 2017-09-11 by twimon

coolchip, 2017-09-12:
Ok. Thanks anyway.