coolchip/node-red-contrib-smartmeter #5: node does not show up

rollercontainer, 2018-07-08:
Hi, 
I installed your package and it shows up in the "manage palette", but the nodes in the left menu aren't there. What went wrong?
node-red was ofc restarted.

rollercontainer, 2018-07-08:
Node-RED version: v0.17.5

rollercontainer, 2018-07-08:
ok, a verbose start told me:
[smartmeter] Error: Could not locate the bindings file. Tried:

coolchip, 2018-07-08:
Oh, that's very old. I searched for it and found this:
https://github.com/node-red/node-red-nodes/issues/354
So, do you use NPM version 3.x or higher?
Have you tried to npm remove and npm install my node?

coolchip, 2018-07-18:
@rollercontainer did that help you?

rollercontainer, 2018-07-19:
Hm, no. I've a second RPI3B+ with fresh stretch, node-red v0.18.7, node-js v8.11.3, npm 6.2.0 with same result. I installed all globaly and it looks like even serialport is not installed correctly. Will dig further on that route.

closed 2018-07-19 by rollercontainer