coolchip/node-red-contrib-smartmeter #6: Error while parsing SML message

Genosse274, 2018-07-19:
Hello, i recieved this error. It ran fine for days. Maybe this helps improving.
`Jul 19 08:59:34 raspberrypi Node-RED[5224]: 19 Jul 08:59:34 - [error] smartmeter-obis error: Error: Error while parsing SML message: Error: Unknown TL-Field for SmlList!: Message: 1b1b1b1b010101017607000900f064236200620072630101760101070009003b21610b0901454d48000067e964010163334d007607000900f06424620062007263070177010b0901454d48000067e964070100620affff72620165003b27ce480177070100000009ff010101010b0901454d48000067e9640177070100020800ff640100a201621e52ff560000e484c30177070100020801ff0101621e52ff560000e484c30177070100020802ff0101621e52ff5600000000000177070100100700ff0101621b52ff55ffffd1450177078181c78205ff0101010183020a13d540a9ef7dc2d6c22214b64d79f3f33c3455ac67606c638e7e038c8cf34a2b616740fa813ff0c5bc2710ab5c3d730101016338e7007607000900f06427620062007263020171016367760000001b1b1b1b1a02fed0
Jul 19 08:59:34 raspberrypi Node-RED[5224]: 19 Jul 08:59:34 - [red] Uncaught Exception:
Jul 19 08:59:34 raspberrypi Node-RED[5224]: 19 Jul 08:59:34 - TypeError: Cannot read property 'logger' of undefined
Jul 19 08:59:34 raspberrypi Node-RED[5224]:     at SerialPort.<anonymous> (/home/username/.node-red/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:223:30)
Jul 19 08:59:34 raspberrypi Node-RED[5224]:     at SerialPort._error (/home/username/.node-red/node_modules/serialport/lib/serialport.js:199:14)
Jul 19 08:59:34 raspberrypi Node-RED[5224]:     at process.nextTick (/home/username/.node-red/node_modules/serialport/lib/serialport.js:206:31)
Jul 19 08:59:34 raspberrypi Node-RED[5224]:     at _combinedTickCallback (internal/process/next_tick.js:131:7)
Jul 19 08:59:34 raspberrypi Node-RED[5224]:     at process._tickCallback (internal/process/next_tick.js:180:9)
Jul 19 08:59:34 raspberrypi systemd[1]: nodered.service: Main process exited, code=exited, status=1/FAILURE
Jul 19 08:59:34 raspberrypi systemd[1]: nodered.service: Unit entered failed state.
Jul 19 08:59:34 raspberrypi systemd[1]: nodered.service: Failed with result 'exit-code'.
Jul 19 08:59:34 raspberrypi systemd[1]: nodered.service: Service hold-off time over, scheduling restart.
Jul 19 08:59:34 raspberrypi systemd[1]: Stopped Node-RED graphical event wiring tool..
Jul 19 08:59:34 raspberrypi systemd[1]: Started Node-RED graphical event wiring tool..
Jul 19 08:59:36 raspberrypi Node-RED[29895]: 19 Jul 08:59:36 - [info]
Jul 19 08:59:36 raspberrypi Node-RED[29895]: Welcome to Node-RED
Jul 19 08:59:36 raspberrypi Node-RED[29895]: ===================
Jul 19 08:59:36 raspberrypi Node-RED[29895]: 19 Jul 08:59:36 - [info] Node-RED version: v0.18.4
Jul 19 08:59:36 raspberrypi Node-RED[29895]: 19 Jul 08:59:36 - [info] Node.js  version: v8.11.2
Jul 19 08:59:36 raspberrypi Node-RED[29895]: 19 Jul 08:59:36 - [info] Linux 4.14.34-v7+ arm LE
Jul 19 08:59:37 raspberrypi Node-RED[29895]: 19 Jul 08:59:37 - [info] Loading palette nodes
`

coolchip, 2018-07-19:
Thanks for your reply. I will this if I can fix it.

coolchip, 2018-07-19:
@Apollon77 Any ideas?

Apollon77, 2018-07-20:
You said it run file for days and then this error?

Are you sure it is not a „a bit got lost“ problem? Please check that the reading hard is placed exact and cable is fine. 

But I check later the message

Genosse274, 2018-07-20:
Its the first time i noticed this error. Its running more than a month (request interval 10 secs). I think a false reading should not lead to a crash of node red. Surely it could be a hardware problem, for example someone touched the wire. In this case, noone was at home.
When i installed the meter and the connection wasnt fine i got debug messages in node red, which is way better than a restart of node red.
By the way, thanks for the implementation. Setting up was very easy.

Apollon77, 2018-07-20:
@coolchip: Can you make sure to have the latest library version as dependency. The "logger" error should have been fixed waaay ago!
1.1.2 is current one

coolchip, 2018-07-21:
@Apollon77 I still use 1.1.2. But perhaps @Genosse274 has an old version. @Genosse274, can you 'npm remove' and 'npm install' my packet again? Or show us the dependency tree of your Node-RED nodes with 'npm list'.

Apollon77, 2018-07-23:
Then we close here and you reopen if you see another crash like this?

Genosse274, 2018-07-23:
Sure. Sorry

closed 2018-07-23 by Genosse274