coolchip/node-red-contrib-smartmeter #11: CRC Fehler

Jemeins, 2020-03-04:
Moin,

ist es möglich im Smartmeter Plugin die CRC checksum zu ignorieren?
Eventuell in der Konfiguration mit einer Checkbox?
Bei meinem Zähler (Holley DTZ541) gibt es Bekannterweise CRC Fehler.

Mit freundlichen Grüßen

coolchip, 2020-05-21:
Ist natürlich möglich. Komme zeitlich im Moment leider nicht zu. Habe aber nichts gegen einen PullRequest. ;)

hacki11, 2020-10-25:
Mit dem Fix werden auch die CRCs vom Holley überprüft!
https://github.com/david-freistrom/OpenSML/pull/12

coolchip, 2020-12-16:
Habe die Option in der Version 0.5.1 hinzugefügt. Versuch das mal...

hacki11, 2020-12-16:
Das Aktualisieren der Dependency müsste reichen, die crc sollte nun gültig sein. Es kommen tatsächlich viele crc Fehler wodurch bei mir viele 0-Werte zustande kamen

coolchip, 2020-12-17:
Funktioniert es nun oder gibt es noch Probleme? Werden anstatt der Checksummenfehler jetzt 0-Werte geschickt?

hacki11, 2020-12-17:
Ich hab mich ungünstig ausgedrückt.

Ich habe selbst diesen Zähler, der einen falschen CRC Algo benutzt.
Ich habe den CRC Algo in OpenSML eingebaut wodurch die Lib die CRCs nun als gültig ausweist.

Ich nutze node-red-contrib-smartmeter nicht, sondern habe das issue hier gefunden und auf den Fix hingewiesen.

Vielleicht kann der Issueersteller einen Test durchführen, ohne die CRC zu ignorieren.

0-Werte bekommt man nur, wenn man die CRC Fehler ignoriert. Das war für mich ein Problem daher die Anpassung von OpenSML.

Jemeins, 2020-12-17:
Vielen Dank für den Fix!
Leider ist es mir aktuell nicht möglich die Anpassung zu testen.
Ich werde es aber so bald wie möglich probieren!

coolchip, 2021-01-02:
Sehr schön. Danke @hacki11 für die Erweiterung in OpenSML. Selbstverständlich beruht die Erweiterung die ich in node-red-contrib-smartmeter eingebaut habe genau darauf. 

Ich schließe den Issue nun erstmal. Wenn es noch Probleme gibt, einfach wieder öffnen.

closed 2021-01-02 by coolchip

Jemeins, 2021-01-03:
Ich wollte es soeben testen, in der NodeRed Palette Verwaltung finde ich jedoch nur die Version 0.3.1
Wie muss ich bei der Installation vorgehen?
Über deine Hilfe wäre ich sehr dankbar!