coolchip/node-red-contrib-smartmeter #12: Connecting to Serial -> Ethernet Converters

mode2k, 2020-04-10:
Hi,
i have a SML Meter connected to a serial -> ethernet converter. Is there a possiblity to connect to this?

For simulating a serial -> ethernet you can use
 ./remserial -p 22000 -s "9600 raw" /dev/ttyUSB_1
on a computer physicaly connected to the smartmeter on /dev/ttyUSB_1. The data is available on TCP 22000.

BR

mode

mode2k, 2020-04-10:
The used smartmeter obis lib supports tcp client. See https://github.com/Apollon77/smartmeter-obis/blob/master/lib/transports/TCPTransport.js

Things to do:
Extend transport configuration by item TCPTransport
Add new configuration objects:
transportTcpHost
transportTcpPort
transportTcpMaxBufferSize
transportTcpMessageTimeout
an pass them to the options object.

mode2k, 2020-04-12:
See my pull request: https://github.com/coolchip/node-red-contrib-smartmeter/pull/14

coolchip, 2020-05-21:
I've merged pull request.

closed 2020-05-21 by coolchip