coolchip/node-red-contrib-smartmeter #18: Optional input from Node

ei23-fxg, 2020-08-15:
Thanks a lot for your work!
I have a request for a feature that i did not find yet.
An input directly into the node where you can feed binary or hex data directly into the node.
That would be convenient.

coolchip, 2020-12-16:
What should happen to the data that is sent into the node?

coolchip, 2020-12-17:
Sorry.... reopen

ei23-fxg, 2020-12-17:
> What should happen to the data that is sent into the node?
Hi!
Ich bin mal so frech und schreibe ich die Antwort auf Deutsch:

Wenn ich ein entferntes Gerät, wie beispielsweise ein Wemos D1, an einem Zähler betreibe und die SML Daten binär über einen UDP-Stream schicke oder über MQTT, dann muss ich die Daten entweder direkt auf dem Chip verarbeiten und ganz ohne den Smartmeter Node arbeiten, oder die Daten zunächst in eine Datei schreiben, um sie dann mit dem Smartmeter Node wieder auszulesen.
Das Auslesen einer Datei ist ja zudem auch über einen File Node möglich. In so fern redundant?!
Hätte der Smartmeter Node einen Input, wäre man an dieser Stelle deutlich flexibler und müsste nicht den Umweg über eine Datei gehen.
Oder übersehe ich da eine Möglichkeit?

ei23-fxg, 2020-12-17:
> What should happen to the data that is sent into the node?

Hi!
Ich bin mal so frech und schreibe ich die Antwort auf Deutsch:

Wenn ich ein entferntes Gerät, wie beispielsweise ein Wemos D1, an einem Zähler betreibe und die SML Daten binär über einen UDP-Stream schicke oder über MQTT, dann muss ich die Daten entweder direkt auf dem Chip verarbeiten und ganz ohne den Smartmeter Node arbeiten, oder die Daten zunächst in eine Datei schreiben, um sie dann mit dem Smartmeter Node wieder auszulesen.
Das Auslesen einer Datei ist ja zudem auch über einen File Node möglich und in so fern redundant?!
Von deinem TCP-Socket lesen geht zwar auch, allerdings hatte ich da öfter Verbindungsabbrüche und ich musste dann den Node bzw. NodeRED neu starten.
Hätte der Smartmeter Node einen Input, wäre man an dieser Stelle deutlich flexibler und müsste nicht den Umweg über eine Datei gehen.
Oder übersehe ich da eine Möglichkeit?

coolchip, 2021-01-02:
Ah okay. Ich verstehe. Theroretisch könnte auch der Wemos D1 als Server fungieren und die Daten, die er ausließt über eine Web-API anbieten. Aber es ginge natürlich auch, wenn man einen Eingang an den Node macht.