coolchip/node-red-contrib-smartmeter #20: Transport Parameter

Peter54879, 2020-09-04:
Hallo,

sobald ich die Transport Parameter einstelle und per deploy Button die Einstellungen aktivieren möchte ist der Node-Red Server nicht mehr erreichbar. Ich muss dann die kompletten Flows über die Console löschen damit ich Node-Red wieder starten kann. Ist so ein Problem bekannt? Ich nutzte Node-Red v0.20.7 und Node.js v8.16.0 auf einem IoT2040 von Siemens.

Danke!

Gruß Peter

coolchip, 2020-12-10:
Nein, ist nicht bekannt. Was genau hast du einstellt? Dann kann ich versuchen das nachzustellen.

Peter54879, 2020-12-13:
Ich habe das Problem lösen können. Ich musste den vorinstallierten Serial Node komplett entfernen und dann alles neu installieren müssen. Jetzt funktioniert alles wie gewünscht. Danke für die Rückmeldung.

coolchip, 2020-12-16:
Danke für die Rückmeldung

closed 2020-12-16 by coolchip