coolchip/node-red-contrib-smartmeter #23: Installation Error

webman2030, 2020-12-15:
Hi I get this Error in Node Red

`2020-12-15T06:50:47.869Z installieren : node-red-contrib-smartmeter 0.3.1

2020-12-15T06:50:48.128Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production node-red-contrib-smartmeter@0.3.1
2020-12-15T06:51:02.379Z [err] npm
2020-12-15T06:51:02.379Z [err]  WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
2020-12-15T06:51:04.173Z [err] npm 
2020-12-15T06:51:04.173Z [err] WARN deprecated har-validator@5.1.5: this library is no longer supported
2020-12-15T06:51:07.844Z [out] 
2020-12-15T06:51:07.844Z [out] > @serialport/bindings@8.0.8 install /config/node-red/node_modules/@serialport/bindings
2020-12-15T06:51:07.844Z [out] > prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2020-12-15T06:51:07.844Z [out] 
2020-12-15T06:51:08.891Z [err] prebuild-install
2020-12-15T06:51:08.891Z [err]  
2020-12-15T06:51:08.892Z [err] WARN
2020-12-15T06:51:08.892Z [err]  
2020-12-15T06:51:08.892Z [err] install
2020-12-15T06:51:08.893Z [err]  No prebuilt binaries found (target=12.18.4 runtime=node arch=arm libc=musl platform=linux)
2020-12-15T06:51:10.794Z [err] gyp
2020-12-15T06:51:10.795Z [err]  
2020-12-15T06:51:10.795Z [err] ERR! build error 
2020-12-15T06:51:10.797Z [err] gyp
2020-12-15T06:51:10.797Z [err]  ERR!
2020-12-15T06:51:10.797Z [err]  stack
2020-12-15T06:51:10.797Z [err]  Error: not found: make
2020-12-15T06:51:10.797Z [err] gyp
2020-12-15T06:51:10.797Z [err]  ERR! 
2020-12-15T06:51:10.798Z [err] stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:13:12)
2020-12-15T06:51:10.798Z [err] gyp
2020-12-15T06:51:10.798Z [err]  ERR! 
2020-12-15T06:51:10.798Z [err] stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:68:19)
2020-12-15T06:51:10.798Z [err] gyp ERR!
2020-12-15T06:51:10.798Z [err]  stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:80:29)
2020-12-15T06:51:10.798Z [err] gyp ERR! 
2020-12-15T06:51:10.799Z [err] stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:89:16
2020-12-15T06:51:10.799Z [err] gyp 
2020-12-15T06:51:10.799Z [err] ERR! stack
2020-12-15T06:51:10.799Z [err]      at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
2020-12-15T06:51:10.799Z [err] gyp 
2020-12-15T06:51:10.799Z [err] ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
2020-12-15T06:51:10.799Z [err] gyp ERR! 
2020-12-15T06:51:10.799Z [err] stack     at FSReqCallback.oncomplete (fs.js:168:21)
2020-12-15T06:51:10.800Z [err] gyp 
2020-12-15T06:51:10.800Z [err] ERR! System Linux 5.4.79-v7l
2020-12-15T06:51:10.801Z [err] gyp ERR! 
2020-12-15T06:51:10.801Z [err] command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
2020-12-15T06:51:10.801Z [err] gyp 
2020-12-15T06:51:10.801Z [err] ERR! cwd /config/node-red/node_modules/@serialport/bindings
2020-12-15T06:51:10.801Z [err] gyp ERR! node -v
2020-12-15T06:51:10.801Z [err]  v12.18.4
2020-12-15T06:51:10.801Z [err] gyp 
2020-12-15T06:51:10.801Z [err] ERR! node-gyp -v v5.1.0
2020-12-15T06:51:10.802Z [err] gyp ERR! not ok
2020-12-15T06:51:10.802Z [err]  
2020-12-15T06:51:11.499Z [err] npm 
2020-12-15T06:51:11.499Z [err] ERR! code ELIFECYCLE
2020-12-15T06:51:11.499Z [err] npm ERR! 
2020-12-15T06:51:11.499Z [err] errno 1
2020-12-15T06:51:11.510Z [err] npm
2020-12-15T06:51:11.510Z [err]  ERR! @serialport/bindings@8.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
2020-12-15T06:51:11.510Z [err] npm ERR!
2020-12-15T06:51:11.510Z [err]  Exit status 1
2020-12-15T06:51:11.510Z [err] npm
2020-12-15T06:51:11.511Z [err]  ERR! 
2020-12-15T06:51:11.511Z [err] npm ERR!
2020-12-15T06:51:11.511Z [err]  Failed at the @serialport/bindings@8.0.8 install script.
2020-12-15T06:51:11.511Z [err] npm
2020-12-15T06:51:11.511Z [err]  ERR! This is probably not a problem with npm. There is likely additional logging output above.
2020-12-15T06:51:11.542Z [err] 
2020-12-15T06:51:11.542Z [err] npm ERR! A complete log of this run can be found in:
2020-12-15T06:51:11.542Z [err] npm ERR!
2020-12-15T06:51:11.542Z [err]      /root/.npm/_logs/2020-12-15T06_51_11_523Z-debug.log`

coolchip, 2020-12-15:
Hello,
you need the linux build tools. This depends on your ditro. If you using an debian based distro (for example Raspbian), please install:
`sudo apt-get install build-essential`

webman2030, 2020-12-15:
Hi,

i use Hass.io (Home Assistant in Container)
there is no way to install it, is there another chance?

coolchip, 2020-12-15:
The container is based on Linux. It should be possible to install the build tools in the container.
Perhaps this helps:
https://github.com/home-assistant/hassio-build
https://community.home-assistant.io/t/how-to-install-make-tools-in-hass-io-node-red-add-on/170566

ei23-fxg, 2020-12-17:
> 
> 
> Hi,
> 
> i use Hass.io (Home Assistant in Container)
> there is no way to install it, is there another chance?

If you are using a Raspberry Pi you could have a look at my smart home project / installation script:
https://ei23.de/diy-smarthome/ [GERMAN]
https://ei23.com/diy-smarthome/ [ENGLISH]

It's designed to be an alternative to the Hassio supervised installer. You get more much more flexibility especially with NodeRED here.

closed 2021-01-02 by coolchip