coolchip/node-red-contrib-smartmeter #25: Hardware for the node

maru2648, 2021-01-13:
First of all, thank you for your excellent work. 
I have a question about the hardware setup. Currently I am working on a university project and here I have to build my own smart meter for only one component, which writes the consumption data into a CSV file.  Thereby I came across your nodes. I have already received an electricity meter from my university https://www.amazon.de/Tech-SDM230Modbus-Wechselstromz%C3%A4hler-Stromz%C3%A4hler-Schnittstelle/dp/B01N8XMS2J. Now I wanted to ask if this is suitable for the node. The communication with the Raspberry Pi I would do via an RS485 adapter https://www.amazon.de/USB-RS485-Adapter-mit-Geh%C3%A4use/dp/B00I9H5J02. 

Is the setup possible or do I have to rethink and possibly look for another electricity meter?

Best regards, 
maru2648

coolchip, 2021-01-13:
Hallo,

you're welcome.
My node is based on that library:
https://github.com/Apollon77/smartmeter-obis
So if you want to know whether it's working or not you have to look at this
lib.

But it seems that your meter has a simple Modbus connection. So use Modbus
instead. There is a very good Modbus node available for Node-RED:
https://flows.nodered.org/node/node-red-contrib-modbus

Best regards,
Sebastian

maru2648, 2021-01-13:
Thanks for your fast response :)

closed 2021-01-13 by maru2648