coolchip/node-red-contrib-smartmeter #26: Engelmann SensoStar U meter support?

FreeTechNick, 2021-01-30:
Hi,
my "district heating provider" has replaced the current meter with an Engelmann SensoStar U. Unfortunately, I can not manage to read the data from the meter. Is this supported or does anyone have experience with it? Thank you.

Regards Markus

coolchip, 2021-02-01:
Hello. This node is based on the smartmeter-obis module. You have to ask the mentainer of this module to add support for this. Than my node has the ability to do it.

You can find the source here: https://github.com/Apollon77/smartmeter-obis

If smartmeter-obis is supporting it but you miss an configuration parameter, come back to me. ;-)

coolchip, 2021-11-28:
If I can do anything else, please ask.

closed 2021-11-28 by coolchip