coolchip/node-red-contrib-smartmeter #27: Installation 0.5.1

thoduh, 2021-03-09:
node-red-contrib-smartmeter lässt sich nicht von 0.3.1 auf 0.5.1 updaten.
Version 0.3.1 hatte ich ohne Probleme lange in Verwendung.
Auch ein entfernen und erneutes installieren von node-red-contrib-smartmeter 0.5.1 funktioniert nicht.
Woran kann das liegen?

[image]

coolchip, 2021-03-15:
Hallo,
ich weiß das hilft nicht viel weiter, aber bei mir geht es. Wenn ich die Fehler in der Ausgabe sehe, dann fehlt hier wohl Python und ich nehme an, dass auch sämtliche anderen Build-Tools fehlen. Die werden benötigt, wenn ein native-node-module installiert werden soll. Das habe ich leider nicht in der Hand, da dieses Modul auch nur ein Wrapper um ein Modul ist, welches ich gar nicht geschrieben habe. 
Um welches Gerät handelt es sich denn? Ich nehme an ein Raspberry Pi?

thoduh, 2021-03-16:
Also über die letzten Monate lief alles Problemlos, nur smartmeter ließ sich nicht updaten. Hab es daraufhin gelöscht und jetzt bekomm ich es gar nicht mehr installiert.
Das plötzlich Python fehlt kann doch fast nicht sein, lief ja bisher auch so.
Sehr seltsam...
Ach ja, es läuft auf einem Raspberry Pi.

Irgend einen Tipp was ich jetzt tun kann damit smartmeter wieder läuft?

thoduh, 2021-03-16:
Hat sich wohl erledigt, durch ein Update von RedMatic auf V 7.2.1 läuft jetzt smartmeter wieder :-)

closed 2021-03-16 by thoduh