coolchip/node-red-contrib-smartmeter #28: smartmeter-obis error: Error: No or too long answer from Serial Device...

JE3001, 2021-04-27:
I've tried to read my electrical smart meter with node red via the contrib-smartmeter node. The node outputs the following error massage:
**"smartmeter-obis error: Error: No or too long answer from Serial Device after last request."**

I've checked the IR interface with Minicom and got some cryptic inputs (see attached screen shot)
[image]
[image]

, so my assumption is that the interface works fine. I've also attached the settings of the node, node red version, etc.
[image]
[image]

I use following HW:
- Raspberry PI 1B 
- IR phototransistor connected to RXD (Pin 10) with pull up
- eHZ Generation K smartmeter (EMH metering GmbH & Co. KG) 

Can anybody help here.

P1Rebo, 2021-12-04:
Hi @coolchip, 
I had this up and running 2 years ago, it was doing pretty fine but needed to cancel the installation. 
As I tried to install everything a RPI4 + newest Node Red docker image I'm getting the same error.

Thanks for your help, P1Rebo

coolchip, 2021-12-17:
@JE3001 I've added a config parameter "debugging" in v0.6.0. This enables you to put smartmeter-obis in debug mode. You are then able to send HEX output to the smartmeter-obis maintainer Apollon77.

coolchip, 2021-12-17:
@P1Rebo Please also use debug mode and ask Apollon77.

closed 2021-12-17 by coolchip

Graefer, 2022-02-11:
Anything new concerning this topic? I get these messages, too.

coolchip, 2022-02-13:
Sorry. I cannot you help with this. I only use the lib from Apollon77. For me the node is still working fine and delivers me the data from my smart meter. I use Node v12.x and Node-RED 2.1.3. 

If you get "No or too long answer" then enable debugging and post your HEX data. If no hex data is received check your signal source.
If you use Node-RED in a Docker container be sure the docker container can use the hardware ressource (IR Reader) from your host.

Graefer, 2022-03-01:
I could solve the problem by changing the USB plug to another port. Strange...

coolchip, 2022-03-02:
Fine. I also had curious problems for a very long time with several usb to serial converters. Changing the power source from my SBC does the trick. Since I use the new one the problems are gone.