coolchip/node-red-contrib-smartmeter #29: Fehler bei node-red-node-serialport nach Install von Smartmeter

guenter49xx, 2021-07-08:
Hallo Coolchip,
die node-red-contrib-smartmeter erscheint mir sehr nützlich. Kann die sicher in meinem Projekt gut einsetzen.
Leider bekomme ich aber Probleme im Zusammenspiel mit:
node-red-node-serialport 0.14.1
In dieser Node ist die Suche nach einem Port nicht mehr möglich. Auch die Eingabe eines Ports ohne Suche ließ diese Node nicht weiterarbeiten. 
Ich weiß nicht, ob eine Suche des seriellen Ports bei der Smartmeter Node vorgesehen ist; funktioniert jedenfalls auch nicht.
Ich möchte mit einem Adpter meinen Zähler über die Infrarotschnittstelle lesen. Der Adapter liefert ein TTL Signal mit dem SML-Protokoll. Damit sollte doch eine Anbindung an die Smartmeter-Node seriell möglich sein. 
Die serielle Schnittstelle wird in meinem Projekt zwingend für andere Anbindungen benötigt.
Gruß
günter49xx

coolchip, 2021-07-10:
Hallo,
weiß schon gar nicht mehr ob oder wie ich die Suche implementiert habe.
Unter welchen Betriebssystem verwendest du das denn und wo landet dein
Adapter im System?

Schöne Grüße, Sebastian

Napfton, 2021-10-31:
Hab das gleiche Problem nach einem größerem Sys Upgrade auch inkl neuem NodeJS.
smartmeter-obi unterstützt angeblich wegen serialport nur NodeJS 4, 6 und 8 hab ich das richtig verstanden? Laut dderen Abhängigkeiten brauchen die (und damit auch smartmeter hier) mindestens 9.2.4 was es unter NodeJS 8 schon gibt bzw sich bauen lässt.

Wenn ich einen normalen Serial Port (aus der Palette) nehme dann sehe ich die Binärdaten ohne Probleme.
Wenn ich in node-red-contrib-smartmeter SML als protokoll wähle kommt das er das XML nicht lesen kann (ist ja auch D0)
Wenn ich in node-red-contrib-smartmeter D0 auswähle kommt ein smartmeter-obi error, das keine Nachricht vom Device kam oder diese zu lang war.

GIbt es jemanden der 0.5.1 mit aktuellem NodeJS am laufen hat? Oder gibt es eine Möglichkeit den node SerialPort aus der Palette zu verwenden?

Napfton, 2021-10-31:
Achso und wenn ich auf das Such Symbol für die Serielle Schnittstelle klicke so wie @guenter49xx dann kommt das hier:

31 Oct 14:09:28 - TypeError: SerialPort.list no longer takes a callback and only returns a promise
    at Function.SerialPort.list (/home/pi/.node-red/node_modules/@serialport/stream/lib/index.js:651:11)
    at /home/pi/.node-red/node_modules/node-red-contrib-smartmeter/smartmeter-connection.js:21:20

Und node-red steigt aus.

coolchip, 2021-11-28:
Ich habe mal die Paketabhänigkeiten aktualisiert. Vielleicht es dadurch besser geworden.
Welche Node.js Version verwendet ihr? Ich bin bei 14.x

kj187, 2021-11-29:
Ich schliesse mich mal an.

Im Einsatz sind: 
Node-RED: 2.1.3
NodeJS: 14.18.1
Smartmeter: 0.5.1

Wenn ich die Smartmeter Node hinzufüge kommt folgender Error im Log. Die Suche nach dem Serial Port ist ebenfalls nicht Möglich, das Such Icon ist nicht klickbar. Trage ich den Serial Port per Hand ein und drücke auf Speichern schmiert die komplette Node-RED instanz ab und ich muss Node-RED neustarten. 

```
(node:3189) UnhandledPromiseRejectionWarning: TypeError: SerialPort.list no longer takes a callback and only returns a promise
    at Function.SerialPort.list (/home/pi/.node-red/node_modules/@serialport/stream/lib/index.js:651:11)
    at /home/pi/.node-red/node_modules/node-red-contrib-smartmeter/smartmeter-connection.js:21:20
    at Layer.handle [as handle_request] (/usr/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
    at next (/usr/lib/node_modules/node-red/node_modules/express/lib/router/route.js:137:13)
    at /usr/lib/node_modules/node-red/node_modules/@node-red/editor-api/lib/auth/index.js:75:13
    at Layer.handle [as handle_request] (/usr/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
    at next (/usr/lib/node_modules/node-red/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/usr/lib/node_modules/node-red/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/usr/lib/node_modules/node-red/node_modules/express/lib/router/layer.js:95:5)
    at /usr/lib/node_modules/node-red/node_modules/express/lib/router/index.js:281:22
(Use `node --trace-warnings ...` to show where the warning was created)
(node:3189) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:3189) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

kj187, 2021-11-29:
Ah ich seh grad das in Node-RED in der Modul Palette noch die Version 0.5.1 nur angeboten wird, du hast gestern die 0.5.2 veröffentlicht. Mal sehen wie ich die in Node-RED bekomme. Ich melde mich wieder wenn ich die 0.5.2 installiert hab

kj187, 2021-11-29:
Update: mit der 0.2.5 

Der Fehler oben ist geblieben, die Suche nach dem Seriellen Port geht weiterhin nicht. Aber ich kann diesen jetzt ohne Problemlos manuell eintragen und das ganze deployen ohne das Node-RED abschmiert. Von daher, hat mir das Update auf 0.5.2 geholfen. Danke

coolchip, 2021-12-17:
Habe die Port-Auswahl jetzt repariert. Ist in v0.5.3 veröffentlicht.

closed 2021-12-17 by coolchip