coolchip/node-red-contrib-smartmeter #30: No installation unter Node-Red 2.1.3

CarJu, 2021-11-27:
I get reproducable errors, when I try to install your nodes under a new Node-Red installation.

Your Node is so cool und usefull and I really, really hope that you can fix that.

Here is the log from installation:

27 Nov 11:00:54 - [info] Installing module: node-red-contrib-smartmeter, version: 0.5.1
27 Nov 11:01:17 - [warn] Installation of module node-red-contrib-smartmeter failed:
27 Nov 11:01:17 - [warn] ------------------------------------------
27 Nov 11:01:17 - [warn] npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code 1
npm ERR! path /config/node-red/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! prebuild-install WARN install No prebuilt binaries found (target=14.18.1 runtime=node arch=arm64 libc=musl platform=linux)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@14.18.1 | linux | arm64
npm ERR! gyp info find Python using Python version 3.9.5 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/config/node-red/node_modules/@serialport/bindings/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/root/.cache/node-gyp/14.18.1/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/14.18.1',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/14.18.1/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/config/node-red/node_modules/@serialport/bindings',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: not found: make
npm ERR! gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:10:17)
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:57:18
npm ERR! gyp ERR! stack     at new Promise (<anonymous>)
npm ERR! gyp ERR! stack     at step (/usr/lib/node_modules/npm/node_modules/which/which.js:54:21)
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:71:22
npm ERR! gyp ERR! stack     at new Promise (<anonymous>)
npm ERR! gyp ERR! stack     at subStep (/usr/lib/node_modules/npm/node_modules/which/which.js:69:33)
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:80:22
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
npm ERR! gyp ERR! System Linux 5.10.17-v8
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /config/node-red/node_modules/@serialport/bindings
npm ERR! gyp ERR! node -v v14.18.1
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-11-27T10_01_17_125Z-debug.log
27 Nov 11:01:17 - [warn] ------------------------------------------
Error: Install failed

coolchip, 2021-11-28:
Please try it again with 0.5.2. I've upgeraded all dependencies.

coolchip, 2021-11-28:
I use node.js v14.x and I'm able to install v0.5.2 under node-red v2.1.3.

kds69, 2021-12-01:
sorry for my ignorance, but in the version available in dashboard palette manager v2.1.3 is still 0.5.1. Do I miss something?

KlausHans, 2021-12-05:
Got the same problem and can confirm that in palette manager it is still on 0.5.1. Any help is appreciated.

closed 2021-12-16 by coolchip