coolchip/node-red-contrib-smartmeter #31: Node-Red Home Assistant Add-On error

letmoc, 2021-12-19:
Hello, I get the following error messages when installing. Node-Red is integrated into Home Assistant via an add-on.
Can someone help me?

`2021-12-19T12:20:58.897Z Installieren : node-red-contrib-smartmeter 0.6.0

2021-12-19T12:20:58.997Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.0
2021-12-19T12:21:10.110Z [err] npm
2021-12-19T12:21:10.110Z [err]  WARN deprecated har-validator@5.1.5: this library is no longer supported
2021-12-19T12:21:10.748Z [err] npm
2021-12-19T12:21:10.748Z [err]  WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
2021-12-19T12:21:10.850Z [err] npm WARN
2021-12-19T12:21:10.850Z [err]  deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
2021-12-19T12:21:14.588Z [err] npm ERR! 
2021-12-19T12:21:14.589Z [err] code 1
2021-12-19T12:21:14.589Z [err] npm ERR! path /config/node-red/node_modules/@serialport/bindings
2021-12-19T12:21:14.597Z [err] npm ERR! command failed
2021-12-19T12:21:14.597Z [err] npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2021-12-19T12:21:14.597Z [err] npm ERR! prebuild-install WARN install No prebuilt binaries found (target=16.13.1 runtime=node arch=arm64 libc=musl platform=linux)
2021-12-19T12:21:14.598Z [err] npm ERR! gyp info it worked if it ends with ok
2021-12-19T12:21:14.598Z [err] npm ERR!
2021-12-19T12:21:14.598Z [err]  gyp info using node-gyp@8.3.0
2021-12-19T12:21:14.598Z [err] npm ERR! gyp info using node@16.13.1 | linux | arm64
2021-12-19T12:21:14.598Z [err] npm 
2021-12-19T12:21:14.598Z [err] ERR! gyp info find Python using Python version 3.9.7 found at "/usr/bin/python3"
2021-12-19T12:21:14.598Z [err] npm ERR! gyp info spawn /usr/bin/python3
2021-12-19T12:21:14.598Z [err] npm
2021-12-19T12:21:14.598Z [err]  ERR! gyp info spawn args [
2021-12-19T12:21:14.598Z [err] npm ERR!
2021-12-19T12:21:14.598Z [err]  gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
2021-12-19T12:21:14.598Z [err] npm ERR! gyp info spawn args   'binding.gyp',
2021-12-19T12:21:14.598Z [err] npm 
2021-12-19T12:21:14.599Z [err] ERR! gyp info spawn args   '-f',
2021-12-19T12:21:14.599Z [err] npm ERR! gyp info spawn args   'make',
2021-12-19T12:21:14.599Z [err] npm
2021-12-19T12:21:14.599Z [err]  ERR! gyp info spawn args   '-I',
2021-12-19T12:21:14.599Z [err] npm ERR! gyp info spawn args   '/config/node-red/node_modules/@serialport/bindings/build/config.gypi',
2021-12-19T12:21:14.599Z [err] npm ERR! gyp info spawn args   '-I',
2021-12-19T12:21:14.599Z [err] npm ERR!
2021-12-19T12:21:14.599Z [err]  gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
2021-12-19T12:21:14.599Z [err] npm ERR! gyp info spawn args   '-I',
2021-12-19T12:21:14.599Z [err] npm
2021-12-19T12:21:14.599Z [err]  ERR! gyp info spawn args   '/root/.cache/node-gyp/16.13.1/include/node/common.gypi',
2021-12-19T12:21:14.599Z [err] npm ERR!
2021-12-19T12:21:14.600Z [err]  gyp info spawn args   '-Dlibrary=shared_library',
2021-12-19T12:21:14.600Z [err] npm ERR! gyp info spawn args   '-Dvisibility=default',
2021-12-19T12:21:14.600Z [err] npm ERR! gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/16.13.1',
2021-12-19T12:21:14.600Z [err] npm ERR!
2021-12-19T12:21:14.600Z [err]  gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
2021-12-19T12:21:14.600Z [err] npm ERR! gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/16.13.1/<(target_arch)/node.lib',
2021-12-19T12:21:14.600Z [err] npm ERR! gyp info spawn args   '-Dmodule_root_dir=/config/node-red/node_modules/@serialport/bindings',
2021-12-19T12:21:14.600Z [err] npm
2021-12-19T12:21:14.600Z [err]  ERR! gyp info spawn args   '-Dnode_engine=v8',
2021-12-19T12:21:14.600Z [err] npm ERR! gyp info spawn args   '--depth=.',
2021-12-19T12:21:14.600Z [err] npm 
2021-12-19T12:21:14.600Z [err] ERR! gyp info spawn args   '--no-parallel',
2021-12-19T12:21:14.600Z [err] npm ERR! gyp info spawn args   '--generator-output',
2021-12-19T12:21:14.600Z [err] npm
2021-12-19T12:21:14.601Z [err]  ERR! gyp info spawn args   'build',
2021-12-19T12:21:14.601Z [err] npm ERR! gyp info spawn args   '-Goutput_dir=.'
2021-12-19T12:21:14.601Z [err] npm ERR! gyp info spawn args ]
2021-12-19T12:21:14.601Z [err] npm ERR! gyp ERR! build error 
2021-12-19T12:21:14.601Z [err] npm ERR! gyp ERR! stack Error: not found: make
2021-12-19T12:21:14.601Z [err] npm ERR! gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:10:17)
2021-12-19T12:21:14.601Z [err] npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:57:18
2021-12-19T12:21:14.602Z [err] npm ERR! gyp ERR! stack     at new Promise (<anonymous>)
2021-12-19T12:21:14.602Z [err] npm ERR! gyp ERR! stack     at step (/usr/lib/node_modules/npm/node_modules/which/which.js:54:21)
2021-12-19T12:21:14.602Z [err] npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:71:22
2021-12-19T12:21:14.602Z [err] npm ERR!
2021-12-19T12:21:14.602Z [err]  gyp ERR! stack     at new Promise (<anonymous>)
2021-12-19T12:21:14.602Z [err] npm ERR! gyp ERR! stack     at subStep (/usr/lib/node_modules/npm/node_modules/which/which.js:69:33)
2021-12-19T12:21:14.602Z [err] npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:80:22
2021-12-19T12:21:14.602Z [err] npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
2021-12-19T12:21:14.603Z [err] npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
2021-12-19T12:21:14.603Z [err] npm ERR! gyp ERR! System Linux 5.10.63-v8
2021-12-19T12:21:14.603Z [err] npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
2021-12-19T12:21:14.603Z [err] npm ERR! gyp ERR! cwd /config/node-red/node_modules/@serialport/bindings
2021-12-19T12:21:14.603Z [err] npm ERR! gyp ERR! node -v v16.13.1
2021-12-19T12:21:14.603Z [err] npm ERR! gyp ERR! node-gyp -v v8.3.0
2021-12-19T12:21:14.603Z [err] npm ERR! gyp ERR! not ok
2021-12-19T12:21:14.762Z [err] 
2021-12-19T12:21:14.763Z [err] npm ERR! A complete log of this run can be found in:
2021-12-19T12:21:14.763Z [err] npm ERR!     /root/.npm/_logs/2021-12-19T12_21_14_611Z-debug.log
2021-12-19T12:21:14.793Z rc=1`

coolchip, 2021-12-19:
That looks very, very scary. Every node-gyp output is handled as an error.

This is not a problem with node-red-contrib-smartmeter but with the execution of node-gyp. Node-gyp is used by npm to compile native npm modules. In this case its the 'serialport' module. This is common on ARM based boards. 

But it seems that the build-tool 'make' is missing. Assuming you are using a Debian based distro open a shell and try:
```bash
sudo apt install build-essential
```
Then try to reinstall node-red-contrib-smartmeter.

coolchip, 2021-12-20:
@letmoc Did it work now?

letmoc, 2021-12-20:
Thank you for your support. The home assistant image is the problem. Not easy. I haven't made it until now

letmoc, 2021-12-20:
Node-RED->
Optionen->
system_packages:

Here you can transfer options in the configuration in Home Assistant. My tests have so far been unsuccessful.

coolchip, 2022-01-04:
Sorry I'm not very familiar with Home Assistant but it seems that the build-essentials are missing. You have to install it using a linux terminal.

letmoc, 2022-01-04:
Hi, I changed my installation and am now using Docker. everything works here. it's better that way

coolchip, 2022-01-05:
I am very pleased

closed 2022-01-05 by coolchip

ce3a, 2022-10-18:
> Hi, I changed my installation and am now using Docker. everything works here. it's better that way

@letmoc I am having the same issue, `node-red-contrib-smartmeter` not working in node-red on home assistant.
What was your fix? Do you have smartmeter working in home assistant?

ce3a, 2022-10-18:
Adding `node-red-contrib-smartmeter` to the `npm_packages` list in node-red add-on configuration resolved the issue.

```
npm_packages:
  - node-red-contrib-smartmeter
```