coolchip/node-red-contrib-smartmeter #32: Real power with sign?

LuFo27, 2021-12-21:
Hello!
In object objName 1-0:15.7.0*255 the current power is shown.
If I generate more power with my PV than I use, I cannot tell.
How can I manage to show the current power with a sign?

coolchip, 2022-01-04:
Thats curious. I've also own a PV system and I get a negative sign. Do you use the SML protocol?

LuFo27, 2022-01-10:
yes, i use sml protocol. Perhaps depending on the meter used?

rschueler, 2022-02-23:
I have the same problem and found following information: 

https://www.mail-archive.com/volkszaehler-dev@demo.volkszaehler.org/msg02161.html
It says, that there is a bit in message 1.8.0 which indicates the sign of message 15.7.0 value.

But I don´t have a clue how to access this bit or maybe it is not even parsed and exported to the json object.

coolchip, 2022-04-13:
You can put the node in Debug-Mode and send the data to the guy, who wrotes this:
https://github.com/Apollon77/smartmeter-obis
My node is based on this module.

StofLE, 2022-09-21:
Same problem :-(
Interested in reading the sign (direction) of power.

@coolchip: I activated the debug mode. But where is the output?
I would like to send it to Apollon77.

Stefan

jaeger-audiodesign, 2023-04-21:
Did anyone find a solution for this? The smart meter which my father has, already only has the 15.7. with the energy without a sign.