coolchip/node-red-contrib-smartmeter #35: extremly large log file

Stasfe, 2022-03-15:
Hello, 
I'm using this node to read my easymeter, but after two day I could'nt boot my raspberry anymore, cause I was out of space on the sd card. After I managed to open the files on my sd card on other linux machine, I saw the huge deamon.log files that filled my raspberry. 

The log file is full with this information. Like 1000 entrys for 1 sec. 

`Mar 13 00:00:02 raspberrypi Node-RED[351]: NEW DATA
Mar 13 00:00:02 raspberrypi Node-RED[351]: CURRENT PROCESS STEP 4 IN CHECKMESSAGE
Mar 13 00:00:02 raspberrypi Node-RED[351]: MATCH-RESULT DATA: "\r\n1-0:0.0.0*255(0273260481404)\r\n1-0:1.8.0*255(00012911.0592
550*kWh)\r\n1-0:21.7.0*255(000093.80*W)\r\n1-0:41.7.0*255(000014.45*W)\r\n1-0:61.7.0*255(000012.70*W)\r\n1-0:1.7.0*255(000120.
95*W)\r\n1-0:96.5.5*255(80)\r\n0" -> null
Mar 13 00:00:02 raspberrypi Node-RED[351]: NEW DATA
Mar 13 00:00:02 raspberrypi Node-RED[351]: CURRENT PROCESS STEP 4 IN CHECKMESSAGE
Mar 13 00:00:02 raspberrypi Node-RED[351]: MATCH-RESULT DATA: "\r\n1-0:0.0.0*255(0273260481404)\r\n1-0:1.8.0*255(00012911.0592
550*kWh)\r\n1-0:21.7.0*255(000093.80*W)\r\n1-0:41.7.0*255(000014.45*W)\r\n1-0:61.7.0*255(000012.70*W)\r\n1-0:1.7.0*255(000120.
95*W)\r\n1-0:96.5.5*255(80)\r\n0-" -> null`

how to stop this writing into log files?

juggledad, 2022-03-17:
To follow up from the Node-RED forum, 
1) the time is in seconds
2) when you check off the debug node, it tells the underlining node (smartmeter Module [smartmeter@github](https://github.com/Apollon77/smartmeter-obis)) to do debugging and it does a log. That is what caused your storage to fill up.

@coolchip would you be up to having me create a PR with the following added to the node:
[image]

coolchip, 2022-04-13:
Thank you @juggledad [juggledad](https://github.com/juggledad) for helping

closed 2022-04-13 by coolchip