coolchip/node-red-contrib-smartmeter #36: Trigger input

KlausHans, 2022-03-16:
Hello, thank you for that node, its very nice. 

I would like to propose a small feature: An input to trigger the reading. Currently, the only trigger is the request interval inside the node. I would like to use an external trigger.

Background: I want to trigger the smartmeter and my pv inverter modbus reading at the same time, to get close (in terms of time) values for consistent calculations (to calculate self use of pv power for example). Currently i use the smartmeter node as trigger for the modbus readings. Thats not good in case the smartmeter node has problems, as it happened some weeks ago. In this case also my modbus readings weren't triggered. Thats why i want to use an external trigger/tick generator.

KlausHans, 2022-03-16:
Just saw that someone else also wants this if i understand correctly https://github.com/coolchip/node-red-contrib-smartmeter/issues/33#issuecomment-1000248766 :)

staffler, 2022-03-24:
I have the same problem. I also need an external trigger for this node. Would be grate if you could add this function to the node. Thanks

coolchip, 2022-04-13:
will merge that pr soon.

closed 2022-04-13 by coolchip