coolchip/node-red-contrib-smartmeter #41: DD3 eBZ Protokoll wird nicht unterstützt?

sporttiger1000, 2022-05-07:
Hi, 
ich möchte gerne meinen neuen Zweirichtungszähler (DD3 ebz gmbh) per Node-Red und USB-Interface auslesen. Leider kommt im Debug node nichts an bzw. die Fehlermeldung: 
`smartmeter-obis error: Error: No or too long answer from Serial Device after last request.`

Per Serial Node kommt folgender Binärcode an (in Dezimaldarstellung):
`[175,197,66,90,53,68,68,51,178,210,48,54,68,212,65,95,177,48, ... 

177,170,178,53,53,40,48,48,48,178,177,54,170,235,215,232,169,141,10,177,45,48,58,177,46,184,46,178,170,178,53,53,40,48,48,48,177,177,57,170,235,215,232,169,141,10,177,45,48,58,178,46,184,46,48,170,178,53,53,40,48,48,48,48,180,54,170,235,215,232,169,141,10,177,45,...
48,48,51,65,66,66,197,68,169,141,10,33,141,10]`

Ist das eine neue Version vom SML Protokoll? Warum kann der Input nicht interpretiert werden? Binärcode Teilweise gekürzt.
Danke und Gruß
Valentin

sporttiger1000, 2022-05-09:
Issue kann geschlossen werden, der Zähler benötigte das Format 9600 Baud 7-even-1 anstatt 8-N-1.

coolchip, 2022-08-24:
Danke für die Rückmeldung.