coolchip/node-red-contrib-smartmeter #43: Installation failing in nodered 3.x

CaseyRo, 2022-07-16:
hi there! Had to remove smartmeter in nodered 3.0 because it failed to run, now trying to reinstall and getting these errors:

`-----------------------------------------------------------
2022-07-16T10:54:22.551Z Install : node-red-contrib-smartmeter 0.6.3

2022-07-16T10:54:22.777Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.3
2022-07-16T10:54:24.073Z [err] npm
2022-07-16T10:54:24.074Z [err]  WARN 
2022-07-16T10:54:24.074Z [err] config production Use `--omit=dev` instead.
2022-07-16T10:54:31.696Z [err] npm
2022-07-16T10:54:31.697Z [err]  ERR! code
2022-07-16T10:54:31.697Z [err]  1
2022-07-16T10:54:31.697Z [err] npm ERR! path
2022-07-16T10:54:31.698Z [err]  /data/node_modules/@serialport/bindings
2022-07-16T10:54:31.702Z [err] npm ERR! command failed
2022-07-16T10:54:31.702Z [err] npm ERR! command
2022-07-16T10:54:31.703Z [err]  sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2022-07-16T10:54:31.703Z [err] npm ERR!
2022-07-16T10:54:31.703Z [err]  prebuild-install warn install No prebuilt binaries found (target=16.16.0 runtime=node arch=arm libc=musl platform=linux)
2022-07-16T10:54:31.703Z [err] npm ERR! gyp info it worked if it ends with ok
2022-07-16T10:54:31.703Z [err] npm 
2022-07-16T10:54:31.703Z [err] ERR! gyp info using node-gyp@3.8.0
2022-07-16T10:54:31.703Z [err] npm ERR! gyp info using node@16.16.0 | linux | arm
2022-07-16T10:54:31.703Z [err] npm ERR! gyp ERR! configure error 
2022-07-16T10:54:31.703Z [err] npm ERR! gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
2022-07-16T10:54:31.703Z [err] npm ERR!
2022-07-16T10:54:31.703Z [err]  gyp ERR! stack     at PythonFinder.failNoPython (/data/node_modules/node-gyp/lib/configure.js:484:19)
2022-07-16T10:54:31.703Z [err] npm ERR! gyp ERR! stack     at PythonFinder.<anonymous> (/data/node_modules/node-gyp/lib/configure.js:406:16)
2022-07-16T10:54:31.703Z [err] npm
2022-07-16T10:54:31.704Z [err]  ERR! gyp ERR! stack     at F (/data/node_modules/which/which.js:68:16)
2022-07-16T10:54:31.704Z [err] npm ERR!
2022-07-16T10:54:31.704Z [err]  gyp ERR! stack     at E (/data/node_modules/which/which.js:80:29)
2022-07-16T10:54:31.704Z [err] npm ERR! gyp ERR! stack     at /data/node_modules/which/which.js:89:16
2022-07-16T10:54:31.704Z [err] npm 
2022-07-16T10:54:31.704Z [err] ERR! gyp ERR! stack     at /data/node_modules/isexe/index.js:42:5
2022-07-16T10:54:31.704Z [err] npm ERR! gyp ERR! stack     at /data/node_modules/isexe/mode.js:8:5
2022-07-16T10:54:31.704Z [err] npm ERR! gyp ERR! stack     at FSReqCallback.oncomplete (node:fs:198:21)
2022-07-16T10:54:31.704Z [err] npm ERR!
2022-07-16T10:54:31.704Z [err]  gyp ERR! System Linux 5.10.103-v7l+
2022-07-16T10:54:31.704Z [err] npm ERR! gyp ERR! command "/usr/local/bin/node" "/data/node_modules/.bin/node-gyp" "rebuild"
2022-07-16T10:54:31.704Z [err] npm 
2022-07-16T10:54:31.704Z [err] ERR! gyp ERR! cwd /data/node_modules/@serialport/bindings
2022-07-16T10:54:31.704Z [err] npm ERR! gyp ERR! node -v v16.16.0
2022-07-16T10:54:31.704Z [err] npm
2022-07-16T10:54:31.704Z [err]  ERR! gyp ERR! node-gyp -v v3.8.0
2022-07-16T10:54:31.704Z [err] npm ERR!
2022-07-16T10:54:31.705Z [err]  gyp ERR! not ok
2022-07-16T10:54:31.709Z [err] 
2022-07-16T10:54:31.710Z [err] npm ERR! A complete log of this run can be found in:
2022-07-16T10:54:31.710Z [err] npm ERR!     /data/.npm/_logs/2022-07-16T10_54_23_982Z-debug-0.log
2022-07-16T10:54:31.733Z rc=1`

I assume this has something to do with the removal of python2 from nodered, but not sure 😉

eiguudewie, 2022-07-17:
Hi CaseyRo,

I had the same problem but managed to make it run again by manually removing some remainders in the file tree which uninstalling couldn't delete. Afterwards the reinstall worked fine.

CaseyRo, 2022-07-17:
@eiguudewie can you tell me which you removed exactly? I've resorted to building a debian based nodered image myself, but would rather not 😉

eiguudewie, 2022-07-17:
@CaseyRo 

I'm using a docker image and had the following errors while trying to uninstall via the palette.

```
2022-07-17T08:14:18.380Z Entfernen : node-red-contrib-smartmeter

2022-07-17T08:17:00.135Z npm remove --no-audit --no-update-notifier --no-fund --save node-red-contrib-smartmeter
2022-07-17T08:17:00.750Z [err] npm
2022-07-17T08:17:00.750Z [err]  WARN old lockfile 
2022-07-17T08:17:00.750Z [err] npm WARN old lockfile
2022-07-17T08:17:00.750Z [err]  The package-lock.json file was created with an old version of npm,
2022-07-17T08:17:00.750Z [err] npm 
2022-07-17T08:17:00.750Z [err] WARN old lockfile
2022-07-17T08:17:00.750Z [err]  so supplemental metadata must be fetched from the registry.
2022-07-17T08:17:00.750Z [err] npm
2022-07-17T08:17:00.750Z [err]  WARN old lockfile 
2022-07-17T08:17:00.751Z [err] npm WARN old lockfile This is a one-time fix-up, please be patient...
2022-07-17T08:17:00.751Z [err] npm 
2022-07-17T08:17:00.751Z [err] WARN old lockfile 
2022-07-17T08:30:56.395Z [err] npm
2022-07-17T08:30:56.395Z [err]  WARN cleanup Failed to remove some directories [
2022-07-17T08:30:56.395Z [err] npm WARN cleanup   [
2022-07-17T08:30:56.395Z [err] npm WARN cleanup     '/data/node_modules/.node-red-contrib-smartmeter-BTEUws5U',
2022-07-17T08:30:56.395Z [err] npm WARN cleanup     [Error: ENOTEMPTY: directory not empty, rmdir '/data/node_modules/.node-red-contrib-smartmeter-BTEUws5U/node_modules/@serialport/bindings/build/Release'] {
2022-07-17T08:30:56.395Z [err] npm WARN cleanup       errno: -39,
2022-07-17T08:30:56.395Z [err] npm WARN cleanup       code: 'ENOTEMPTY',
2022-07-17T08:30:56.395Z [err] npm WARN cleanup       syscall: 'rmdir',
2022-07-17T08:30:56.395Z [err] npm WARN cleanup       path: '/data/node_modules/.node-red-contrib-smartmeter-BTEUws5U/node_modules/@serialport/bindings/build/Release'
2022-07-17T08:30:56.395Z [err] npm WARN 
2022-07-17T08:30:56.395Z [err] cleanup     }
2022-07-17T08:30:56.395Z [err] npm WARN
2022-07-17T08:30:56.395Z [err]  cleanup   ]
2022-07-17T08:30:56.395Z [err] npm WARN cleanup ]
2022-07-17T08:30:56.520Z [out] 
2022-07-17T08:30:56.520Z [out] removed 25 packages in 14m
2022-07-17T08:30:56.529Z rc=0
```

So from the console I removed the whole ´.node-red-contrib-smartmeter-BTEUws5U/´ branch. At first it didn't work because a hidden file in the leaf directory always was recreated when removing one instance. A restart of the image helped.

Trying to reinstall then gave another error:
```
2022-07-17T08:46:24.278Z Installieren : node-red-contrib-smartmeter 0.6.3

2022-07-17T08:49:06.039Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.3
2022-07-17T08:49:06.890Z [err] npm
2022-07-17T08:49:06.891Z [err]  WARN config
2022-07-17T08:49:06.891Z [err]  production Use `--omit=dev` instead.
2022-07-17T08:49:34.236Z [err] npm
2022-07-17T08:49:34.236Z [err]  WARN cleanup Failed to remove some directories [
2022-07-17T08:49:34.236Z [err] npm WARN cleanup   [
2022-07-17T08:49:34.236Z [err] npm WARN cleanup     '/data/node_modules/@serialport/.bindings-x0sNnJYY',
2022-07-17T08:49:34.236Z [err] npm WARN cleanup     [Error: ENOTEMPTY: directory not empty, rmdir '/data/node_modules/@serialport/.bindings-x0sNnJYY/build/Release'] {
2022-07-17T08:49:34.236Z [err] npm WARN 
2022-07-17T08:49:34.236Z [err] cleanup       errno: -39,
2022-07-17T08:49:34.236Z [err] npm WARN cleanup       code: 'ENOTEMPTY',
2022-07-17T08:49:34.236Z [err] npm WARN
2022-07-17T08:49:34.236Z [err]  cleanup       syscall: 'rmdir',
2022-07-17T08:49:34.236Z [err] npm WARN cleanup
2022-07-17T08:49:34.236Z [err]        path: '/data/node_modules/@serialport/.bindings-x0sNnJYY/build/Release'
2022-07-17T08:49:34.236Z [err] npm WARN cleanup
2022-07-17T08:49:34.236Z [err]      }
2022-07-17T08:49:34.236Z [err] npm WARN 
2022-07-17T08:49:34.236Z [err] cleanup   ]
2022-07-17T08:49:34.236Z [err] npm WARN cleanup
2022-07-17T08:49:34.236Z [err]  ]
2022-07-17T08:49:34.366Z [out] 
2022-07-17T08:49:34.366Z [out] added 6 packages, removed 39 packages, and changed 21 packages in 27s
2022-07-17T08:49:34.374Z rc=0
```
After also removing the ´@serialport/´ branch my reinstall worked.
Of course I had removed the smartmeter nodes in my flows.

Before trying to reinstall I tried to update according to `npm audit report`, but I don't know if any of that is needed too.

hth

jokakilla, 2022-07-19:
Not sure if this is the same root cause but I had trouble with node-red-contrib-smartmeter when updating to nodered 3.0 too. I'm using the official node-red docker image (https://hub.docker.com/r/nodered/node-red). The smartmeter nodes were deactivated after the update.

Unfortunately I don't have the log message available anymore but it was like: Please recompile the node because it is not compatible. It must have been something to do with the NodeJs version. I also remember something like NODE_MODULE_VERSION 83 and NODE_MODULE_VERSION 93. I guess 3.0 is using Node 16 and node-red-contrib-smartmeter Node 14?

The solution for me was to remove the container and start a new docker container using tag latest-14 instead of latest. This way Node-Red is running on Node 14 which seems compatible again to this project.

Sorry if this is not related to the issue here but I thought it matches the topic. Maybe my solution is useful to someone.

CaseyRo, 2022-07-20:
I've resorted to building the docker image based on debian, since this still includes python (2). It works for me, but not ideal 😀

coolchip, 2022-07-25:
Thanks for this issue and the help. Will try that soon at my Node-RED instance after vacation.

coolchip, 2022-08-24:
For me it is working in Node-RED 3.x.

All the problems listed here seems to either no or wrong python version or missing build environment for native nodes. I use native node modules. Those modues has to be compiled, if no prebuild binaries are provided by the package maintainer. Compiling or building those modules is done automaticly by node-gyp. But node-gyp needs all the build tools (e.g. build-essential on debian). Problem here: The Node-RED Docker container has no build tools.

Feel free to re-open if this is still a problem.

closed 2022-08-24 by coolchip

hvorragend, 2022-09-03:
So it is not possible to use your node in the Home Assistant-NodeRed-Addon (Docker) anymore?

or could you please tell me how get this running? Thank you.

hvorragend, 2022-09-04:
And now with more details. :-)

Node Red v3.0.2
Home Assistant Node Red Addon v13.3.2 

```
-----------------------------------------------------------
2022-09-04T12:54:15.226Z Installieren : node-red-contrib-smartmeter 0.6.3

2022-09-04T12:54:15.167Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.3
2022-09-04T12:54:15.698Z [err] npm
2022-09-04T12:54:15.699Z [err]  
2022-09-04T12:54:15.699Z [err] WARN
2022-09-04T12:54:15.699Z [err]  
2022-09-04T12:54:15.699Z [err] config
2022-09-04T12:54:15.699Z [err]  production Use `--omit=dev` instead.
2022-09-04T12:54:22.182Z [err] npm
2022-09-04T12:54:22.182Z [err]  WARN deprecated har-validator@5.1.5: this library is no longer supported
2022-09-04T12:54:22.307Z [err] npm
2022-09-04T12:54:22.307Z [err]  
2022-09-04T12:54:22.307Z [err] WARN
2022-09-04T12:54:22.307Z [err]  
2022-09-04T12:54:22.307Z [err] deprecated
2022-09-04T12:54:22.307Z [err]  uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
2022-09-04T12:54:22.332Z [err] npm 
2022-09-04T12:54:22.332Z [err] WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
2022-09-04T12:54:23.340Z [err] npm
2022-09-04T12:54:23.341Z [err]  
2022-09-04T12:54:23.341Z [err] ERR!
2022-09-04T12:54:23.341Z [err]  
2022-09-04T12:54:23.341Z [err] code
2022-09-04T12:54:23.341Z [err]  1
2022-09-04T12:54:23.341Z [err] npm
2022-09-04T12:54:23.341Z [err]  
2022-09-04T12:54:23.341Z [err] ERR!
2022-09-04T12:54:23.341Z [err]  
2022-09-04T12:54:23.341Z [err] path
2022-09-04T12:54:23.342Z [err]  /config/node-red/node_modules/@serialport/bindings
2022-09-04T12:54:23.343Z [err] npm
2022-09-04T12:54:23.343Z [err]  
2022-09-04T12:54:23.344Z [err] ERR!
2022-09-04T12:54:23.344Z [err]  command failed
2022-09-04T12:54:23.344Z [err] npm
2022-09-04T12:54:23.344Z [err]  
2022-09-04T12:54:23.344Z [err] ERR!
2022-09-04T12:54:23.344Z [err]  
2022-09-04T12:54:23.344Z [err] command
2022-09-04T12:54:23.344Z [err]  sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2022-09-04T12:54:23.345Z [err] npm
2022-09-04T12:54:23.345Z [err]  
2022-09-04T12:54:23.345Z [err] ERR!
2022-09-04T12:54:23.345Z [err]  prebuild-install warn install No prebuilt binaries found (target=16.16.0 runtime=node arch=x64 libc=musl platform=linux)
2022-09-04T12:54:23.345Z [err] npm
2022-09-04T12:54:23.345Z [err]  
2022-09-04T12:54:23.345Z [err] ERR!
2022-09-04T12:54:23.345Z [err]  gyp info it worked if it ends with ok
2022-09-04T12:54:23.345Z [err] npm
2022-09-04T12:54:23.345Z [err]  
2022-09-04T12:54:23.345Z [err] ERR!
2022-09-04T12:54:23.345Z [err]  gyp info using node-gyp@9.0.0
2022-09-04T12:54:23.346Z [err] npm
2022-09-04T12:54:23.346Z [err]  
2022-09-04T12:54:23.346Z [err] ERR!
2022-09-04T12:54:23.346Z [err]  gyp info using node@16.16.0 | linux | x64
2022-09-04T12:54:23.346Z [err] npm
2022-09-04T12:54:23.346Z [err]  
2022-09-04T12:54:23.346Z [err] ERR!
2022-09-04T12:54:23.346Z [err]  gyp ERR! find Python 
2022-09-04T12:54:23.346Z [err] npm
2022-09-04T12:54:23.346Z [err]  
[... 70 lines cut ...]
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! find Python **********************************************************
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! find Python 
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! configure error 
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack Error: Could not find any Python installation to use
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at PythonFinder.fail (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/find-python.js:330:47)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at PythonFinder.runChecks (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/find-python.js:159:21)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at PythonFinder.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/find-python.js:202:16)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at PythonFinder.execFileCallback (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/find-python.js:294:16)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at exithandler (node:child_process:406:5)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at ChildProcess.errorhandler (node:child_process:418:5)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:527:28)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:289:12)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at onErrorNT (node:internal/child_process:478:16)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! stack     at processTicksAndRejections (node:internal/process/task_queues:83:21)
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! System Linux 5.15.64
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! cwd /config/node-red/node_modules/@serialport/bindings
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! node -v v16.16.0
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! node-gyp -v v9.0.0
2022-09-04T12:54:23.358Z [err] npm ERR! gyp ERR! not ok
2022-09-04T12:54:23.358Z [err] 
2022-09-04T12:54:23.358Z [err] npm ERR! A complete log of this run can be found in:
2022-09-04T12:54:23.358Z [err] npm ERR!     /root/.npm/_logs/2022-09-04T12_54_15_650Z-debug-0.log
2022-09-04T12:54:23.371Z rc=1
```

coolchip, 2022-11-03:
Either Python is missing -> install it. Or the PYTHON environment var is not set:
`gyp ERR! find Python - Set the environment variable PYTHON`