coolchip/node-red-contrib-smartmeter #44: SMLprotocol: "1-0:1.8.x" will be not read

martin8310, 2022-07-29:
Hi,

great work and thanks for the project. Do you know why the Parameter 1-0:1.8.1 and 1-0:1.8.2 will be not sorted out from your tool?
Value from -0:1.8.0*255 works without problems!

Smartmeter is Iskra MT631

regards
martin

coolchip, 2022-08-24:
Hello. What do you mean by "sorted out"?
My module uses the great https://github.com/Apollon77/smartmeter-obis implementation for parsing the smartmeter protocols. IF anything doesn't work its mostly because of this.

martin8310, 2022-09-06:
Hallo, damit meine ich, dass er 1-0:1.8.1 and 1-0:1.8.2 nicht aussortiert / empfängt / filtert.

coolchip, 2022-11-03:
Das ist eine gute Frage. Bei mir werden die ausgegeben. Zeigt dein Smartmeter die Daten an?
Ansonsten kann ich nur empfehlen, Apollon77 zu fragen. Er parsed die Daten.