coolchip/node-red-contrib-smartmeter #47: Installation error - serialport/bindings

roedeer82, 2022-10-20:
I get the following error during installation. Any clue?
npm 8.5.5
node v16.15.0

(Then I installed node-red-node-serialport seperately without any problem. But this didn't solve the problem.)

`2022-10-20T18:52:46.036Z Installieren : node-red-contrib-smartmeter 0.6.3

2022-10-20T18:52:46.051Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.3
2022-10-20T18:53:14.982Z [err] npm
2022-10-20T18:53:14.983Z [err]  
2022-10-20T18:53:14.984Z [err] ERR! code 1
2022-10-20T18:53:14.985Z [err] npm
2022-10-20T18:53:14.986Z [err]  
2022-10-20T18:53:14.986Z [err] ERR!
2022-10-20T18:53:14.986Z [err]  
2022-10-20T18:53:14.987Z [err] path /mnt/dietpi_userdata/node-red/node_modules/@serialport/bindings
2022-10-20T18:53:15.003Z [err] npm
2022-10-20T18:53:15.003Z [err]  
2022-10-20T18:53:15.004Z [err] ERR!
2022-10-20T18:53:15.005Z [err]  command failed
2022-10-20T18:53:15.005Z [err] npm
2022-10-20T18:53:15.006Z [err]  ERR! command
2022-10-20T18:53:15.007Z [err]  sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2022-10-20T18:53:15.008Z [err] npm
2022-10-20T18:53:15.008Z [err]  ERR! make: Entering directory '/mnt/dietpi_userdata/node-red/node_modules/@serialport/bindings/build'
2022-10-20T18:53:15.008Z [err] npm 
2022-10-20T18:53:15.008Z [err] ERR!   CXX(target) Release/obj.target/bindings/src/serialport.o
2022-10-20T18:53:15.008Z [err] npm ERR!
2022-10-20T18:53:15.010Z [err]  make: Leaving directory '/mnt/dietpi_userdata/node-red/node_modules/@serialport/bindings/build'
2022-10-20T18:53:15.011Z [err] npm 
2022-10-20T18:53:15.011Z [err] ERR! prebuild-install warn install No prebuilt binaries found (target=16.15.0 runtime=node arch=arm libc= platform=linux)
2022-10-20T18:53:15.011Z [err] npm ERR! gyp info it worked if it ends with ok
2022-10-20T18:53:15.012Z [err] npm ERR! gyp info using node-gyp@9.0.0
2022-10-20T18:53:15.012Z [err] npm ERR! gyp info using node@16.15.0 | linux | arm
2022-10-20T18:53:15.012Z [err] npm
2022-10-20T18:53:15.012Z [err]  ERR! gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
2022-10-20T18:53:15.012Z [err] npm 
2022-10-20T18:53:15.013Z [err] ERR! gyp info spawn /usr/bin/python3
2022-10-20T18:53:15.013Z [err] npm 
2022-10-20T18:53:15.013Z [err] ERR! gyp info spawn args [
2022-10-20T18:53:15.013Z [err] npm ERR!
2022-10-20T18:53:15.013Z [err]  gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
2022-10-20T18:53:15.013Z [err] npm ERR!
2022-10-20T18:53:15.014Z [err]  gyp info spawn args   'binding.gyp',
2022-10-20T18:53:15.014Z [err] npm ERR!
2022-10-20T18:53:15.014Z [err]  gyp info spawn args   '-f',
2022-10-20T18:53:15.014Z [err] npm ERR!
2022-10-20T18:53:15.014Z [err]  gyp info spawn args   'make',
2022-10-20T18:53:15.014Z [err] npm ERR!
2022-10-20T18:53:15.015Z [err]  gyp info spawn args   '-I',
2022-10-20T18:53:15.015Z [err] npm 
2022-10-20T18:53:15.015Z [err] ERR! gyp info spawn args   '/mnt/dietpi_userdata/node-red/node_modules/@serialport/bindings/build/config.gypi',
2022-10-20T18:53:15.015Z [err] npm ERR!
2022-10-20T18:53:15.015Z [err]  gyp info spawn args   '-I',
2022-10-20T18:53:15.015Z [err] npm ERR!
2022-10-20T18:53:15.015Z [err]  gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
2022-10-20T18:53:15.015Z [err] npm ERR! gyp info spawn args   '-I',
2022-10-20T18:53:15.016Z [err] npm ERR! gyp info spawn args   '/mnt/dietpi_userdata/node-red/.cache/node-gyp/16.15.0/include/node/common.gypi',
2022-10-20T18:53:15.016Z [err] npm
2022-10-20T18:53:15.016Z [err]  ERR! gyp info spawn args   '-Dlibrary=shared_library',
2022-10-20T18:53:15.016Z [err] npm 
2022-10-20T18:53:15.016Z [err] ERR! gyp info spawn args   '-Dvisibility=default',
2022-10-20T18:53:15.016Z [err] npm ERR!
2022-10-20T18:53:15.017Z [err]  gyp info spawn args   '-Dnode_root_dir=/mnt/dietpi_userdata/node-red/.cache/node-gyp/16.15.0',
2022-10-20T18:53:15.017Z [err] npm ERR!
2022-10-20T18:53:15.017Z [err]  gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
2022-10-20T18:53:15.017Z [err] npm ERR! gyp info spawn args   '-Dnode_lib_file=/mnt/dietpi_userdata/node-red/.cache/node-gyp/16.15.0/<(target_arch)/node.lib',
2022-10-20T18:53:15.017Z [err] npm ERR! gyp info spawn args   '-Dmodule_root_dir=/mnt/dietpi_userdata/node-red/node_modules/@serialport/bindings',
2022-10-20T18:53:15.017Z [err] npm
2022-10-20T18:53:15.018Z [err]  ERR! gyp info spawn args   '-Dnode_engine=v8',
2022-10-20T18:53:15.018Z [err] npm 
2022-10-20T18:53:15.018Z [err] ERR! gyp info spawn args   '--depth=.',
2022-10-20T18:53:15.018Z [err] npm ERR!
2022-10-20T18:53:15.018Z [err]  gyp info spawn args   '--no-parallel',
2022-10-20T18:53:15.018Z [err] npm ERR!
2022-10-20T18:53:15.019Z [err]  gyp info spawn args   '--generator-output',
2022-10-20T18:53:15.019Z [err] npm ERR!
2022-10-20T18:53:15.019Z [err]  gyp info spawn args   'build',
2022-10-20T18:53:15.019Z [err] npm ERR! gyp info spawn args   '-Goutput_dir=.'
2022-10-20T18:53:15.019Z [err] npm ERR! gyp info spawn args ]
2022-10-20T18:53:15.020Z [err] npm ERR! gyp info spawn make
2022-10-20T18:53:15.020Z [err] npm
2022-10-20T18:53:15.020Z [err]  ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
2022-10-20T18:53:15.020Z [err] npm
2022-10-20T18:53:15.020Z [err]  ERR! In file included from /usr/include/errno.h:28,
2022-10-20T18:53:15.020Z [err] npm 
2022-10-20T18:53:15.020Z [err] ERR!                  from /mnt/dietpi_userdata/node-red/.cache/node-gyp/16.15.0/include/node/uv/errno.h:25,
2022-10-20T18:53:15.020Z [err] npm ERR!
2022-10-20T18:53:15.021Z [err]                   from /mnt/dietpi_userdata/node-red/.cache/node-gyp/16.15.0/include/node/uv.h:54,
2022-10-20T18:53:15.021Z [err] npm ERR!
2022-10-20T18:53:15.021Z [err]                   from ../../../nan/nan.h:59,
2022-10-20T18:53:15.021Z [err] npm ERR!
2022-10-20T18:53:15.021Z [err]                   from ../src/./serialport.h:13,
2022-10-20T18:53:15.021Z [err] npm ERR!                  from ../src/serialport.cpp:1:
2022-10-20T18:53:15.022Z [err] npm ERR! /usr/include/arm-linux-gnueabihf/bits/errno.h:26:11: fatal error: linux/errno.h: No such file or directory
2022-10-20T18:53:15.022Z [err] npm
2022-10-20T18:53:15.022Z [err]  ERR!  # include <linux/errno.h>
2022-10-20T18:53:15.022Z [err] npm
2022-10-20T18:53:15.022Z [err]  ERR!            ^~~~~~~~~~~~~~~
2022-10-20T18:53:15.022Z [err] npm 
2022-10-20T18:53:15.023Z [err] ERR! compilation terminated.
2022-10-20T18:53:15.023Z [err] npm ERR!
2022-10-20T18:53:15.023Z [err]  make: *** [bindings.target.mk:114: Release/obj.target/bindings/src/serialport.o] Error 1
2022-10-20T18:53:15.023Z [err] npm ERR!
2022-10-20T18:53:15.023Z [err]  gyp ERR! build error 
2022-10-20T18:53:15.023Z [err] npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
2022-10-20T18:53:15.023Z [err] npm
2022-10-20T18:53:15.024Z [err]  ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
2022-10-20T18:53:15.024Z [err] npm ERR!
2022-10-20T18:53:15.024Z [err]  gyp ERR! stack     at ChildProcess.emit (node:events:527:28)
2022-10-20T18:53:15.024Z [err] npm ERR!
2022-10-20T18:53:15.025Z [err]  gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
2022-10-20T18:53:15.025Z [err] npm ERR! gyp ERR! System Linux 5.10.103-v7+
2022-10-20T18:53:15.025Z [err] npm ERR!
2022-10-20T18:53:15.025Z [err]  gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
2022-10-20T18:53:15.025Z [err] npm ERR! gyp ERR! cwd /mnt/dietpi_userdata/node-red/node_modules/@serialport/bindings
2022-10-20T18:53:15.025Z [err] npm 
2022-10-20T18:53:15.025Z [err] ERR! gyp ERR! node -v v16.15.0
2022-10-20T18:53:15.025Z [err] npm ERR!
2022-10-20T18:53:15.026Z [err]  gyp ERR! node-gyp -v v9.0.0
2022-10-20T18:53:15.026Z [err] npm ERR! gyp ERR! not ok
2022-10-20T18:53:15.040Z [err] 
2022-10-20T18:53:15.042Z [err] npm 
2022-10-20T18:53:15.043Z [err] ERR! A complete log of this run can be found in:
2022-10-20T18:53:15.043Z [err] npm ERR!     /mnt/dietpi_userdata/node-red/.npm/_logs/2022-10-20T18_52_48_204Z-debug-0.log
2022-10-20T18:53:15.116Z rc=1`

coolchip, 2022-11-03:
Sounds wired
npm ERR! /usr/include/arm-linux-gnueabihf/bits/errno.h:26:11: fatal error: linux/errno.h: No such file or directory

I don't know how this might happen. Perhaps try another arm-gcc compiler

coolchip, 2022-11-03:
If this is an option try Armbian Linux for your Board. I'm very happy with it.

bkolbe, 2022-11-13:
I have the same error using truenas with the  Dockerimage of nodered. When doing intallation of the moduile i receive this error:

```
-----------------------------------------------------------
2022-11-13T05:48:10.730Z Installieren : node-red-contrib-smartmeter 0.6.3

2022-11-13T05:48:10.817Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.3
2022-11-13T05:48:11.389Z [err] npm
2022-11-13T05:48:11.389Z [err]  WARN config
2022-11-13T05:48:11.389Z [err]  production Use `--omit=dev` instead.
2022-11-13T05:48:19.696Z [err] npm
2022-11-13T05:48:19.696Z [err]  ERR! code 1
2022-11-13T05:48:19.696Z [err] npm ERR! path /data/node_modules/@serialport/bindings
2022-11-13T05:48:19.698Z [err] npm ERR! command failed
2022-11-13T05:48:19.699Z [err] npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2022-11-13T05:48:19.699Z [err] npm ERR! prebuild-install warn install No prebuilt binaries found (target=16.16.0 runtime=node arch=x64 libc=musl platform=linux)
2022-11-13T05:48:19.699Z [err] npm ERR! gyp info it worked if it ends with ok
2022-11-13T05:48:19.699Z [err] npm ERR! gyp info using node-gyp@9.0.0
2022-11-13T05:48:19.699Z [err] npm ERR! gyp info using node@16.16.0 | linux | x64
2022-11-13T05:48:19.699Z [err] npm ERR!
2022-11-13T05:48:19.699Z [err]  gyp info find Python using Python version 3.10.5 found at "/usr/bin/python3"
2022-11-13T05:48:19.699Z [err] npm ERR! gyp WARN install got an error, rolling back install
2022-11-13T05:48:19.699Z [err] npm ERR! gyp ERR! configure error 
2022-11-13T05:48:19.699Z [err] npm ERR! gyp ERR! stack Error: ENOENT: no such file or directory, mkdir '/.cache'
2022-11-13T05:48:19.699Z [err] npm ERR!
2022-11-13T05:48:19.699Z [err]  gyp ERR! System Linux 5.10.142+truenas
2022-11-13T05:48:19.699Z [err] npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
2022-11-13T05:48:19.699Z [err] npm ERR! gyp ERR! cwd /data/node_modules/@serialport/bindings
2022-11-13T05:48:19.699Z [err] npm ERR!
2022-11-13T05:48:19.699Z [err]  gyp ERR! node -v v16.16.0
2022-11-13T05:48:19.699Z [err] npm ERR! gyp ERR! node-gyp -v v9.0.0
2022-11-13T05:48:19.699Z [err] npm ERR! gyp ERR! not ok
2022-11-13T05:48:19.702Z [err] 
2022-11-13T05:48:19.702Z [err] npm ERR! A complete log of this run can be found in:
2022-11-13T05:48:19.702Z [err] npm ERR!     /data/.npm/_logs/2022-11-13T05_48_11_355Z-debug-0.log
2022-11-13T05:48:19.715Z rc=1
```

roedeer82, 2022-11-17:
I switched from dietpi to Raspberry Pi OS - problem solved.

bkolbe, 2022-11-25:
Any Updte on this? Are there some more information i can deliver?

coolchip, 2022-11-28:
The problem is not this node. **Every** node using native npm modules (ex. serialport) needs binarys. If this binary isn't provided by the package maintainer ("npm ERR! prebuild-install warn install No prebuilt binaries found") you have to build this on your own. I know that this is very complicated but unfortunatelly there is no other way.
To build native modules you need:
- python 
- compiler for your processor architecture

This problem exists mostly on ARM based boards. 
If you use a debian based linux distro try to install python3 and build-essential. If you use Node-RED as docker container it is much worser. Don't use the minimal docker containers. And again the container needs python3 and the compiler. If not, log into your container and install this packages.

coolchip, 2022-11-28:
Try to google this problem.
https://github.com/hobbyquaker/node-red-contrib-zigbee/issues/47

Zahnweh, 2022-12-12:
Does it make sense to open an Issue in the Node-RED Add-On Repo or aren't these the droids I'm looking for?

FredFeuerstein22, 2022-12-12:
Done 
https://github.com/hassio-addons/addon-node-red/issues/1509#issue-1491239057

coolchip, 2023-02-08:
On my opinion also the Node-RED Team cannot solve this issue. But look here:
https://nodered.org/docs/getting-started/docker-custom

In the section "Debian based containers" they describe how to build an custom Node-RED Docker image with python and build-essential. That's, what you want.
And again: I know this is complicated and annoying but unfortunatelly I cannot solve this four you.

soeffi, 2023-03-20:
Have the same problem here, with Node-RED running in Docker on ARM.

It works with
nodered/node-red:3.0.2-16

Issue occurs with
nodered/node-red:3.0.2-18