coolchip/node-red-contrib-smartmeter #48: Installation Error Home Assistant

Test-Nerd, 2022-12-11:
-----------------------------------------------------------
2022-12-11T19:54:05.213Z Installieren : node-red-contrib-smartmeter 0.6.3

2022-12-11T19:54:05.342Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.3
2022-12-11T19:54:08.853Z [err] npm
2022-12-11T19:54:08.855Z [err]  WARN 
2022-12-11T19:54:08.855Z [err] config production Use `--omit=dev` instead.
2022-12-11T19:54:31.857Z [err] npm
2022-12-11T19:54:31.858Z [err]  ERR! code 1
2022-12-11T19:54:31.859Z [err] npm ERR!
2022-12-11T19:54:31.859Z [err]  path /config/node-red/node_modules/@serialport/bindings
2022-12-11T19:54:31.865Z [err] npm ERR! command failed
2022-12-11T19:54:31.865Z [err] npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2022-12-11T19:54:31.866Z [err] npm ERR!
2022-12-11T19:54:31.866Z [err]  prebuild-install warn install No prebuilt binaries found (target=16.17.1 runtime=node arch=arm64 libc=musl platform=linux)
2022-12-11T19:54:31.866Z [err] npm ERR! gyp info it worked if it ends with ok
2022-12-11T19:54:31.866Z [err] npm 
2022-12-11T19:54:31.867Z [err] ERR! gyp info using node-gyp@3.8.0
2022-12-11T19:54:31.867Z [err] npm ERR!
2022-12-11T19:54:31.867Z [err]  gyp info using node@16.17.1 | linux | arm64
2022-12-11T19:54:31.867Z [err] npm ERR! gyp ERR! configure error 
2022-12-11T19:54:31.867Z [err] npm ERR! gyp ERR! stack Error: Can't find Python executable "/usr/bin/python3", you can set the PYTHON env variable.
2022-12-11T19:54:31.867Z [err] npm ERR! gyp ERR! stack     at PythonFinder.failNoPython (/config/node-red/node_modules/node-gyp/lib/configure.js:484:19)
2022-12-11T19:54:31.867Z [err] npm ERR!
2022-12-11T19:54:31.868Z [err]  gyp ERR! stack     at PythonFinder.<anonymous> (/config/node-red/node_modules/node-gyp/lib/configure.js:406:16)
2022-12-11T19:54:31.868Z [err] npm ERR! gyp ERR! stack     at F (/config/node-red/node_modules/node-gyp/node_modules/which/which.js:68:16)
2022-12-11T19:54:31.868Z [err] npm ERR! gyp ERR! stack     at E (/config/node-red/node_modules/node-gyp/node_modules/which/which.js:80:29)
2022-12-11T19:54:31.868Z [err] npm ERR! gyp ERR! stack     at /config/node-red/node_modules/node-gyp/node_modules/which/which.js:89:16
2022-12-11T19:54:31.868Z [err] npm ERR!
2022-12-11T19:54:31.868Z [err]  gyp ERR! stack     at /config/node-red/node_modules/isexe/index.js:42:5
2022-12-11T19:54:31.868Z [err] npm ERR! gyp ERR! stack     at /config/node-red/node_modules/isexe/mode.js:8:5
2022-12-11T19:54:31.868Z [err] npm
2022-12-11T19:54:31.868Z [err]  ERR! gyp ERR! stack     at FSReqCallback.oncomplete (node:fs:206:21)
2022-12-11T19:54:31.868Z [err] npm ERR! gyp ERR! System Linux 5.15.61-v8
2022-12-11T19:54:31.868Z [err] npm ERR!
2022-12-11T19:54:31.869Z [err]  gyp ERR! command "/usr/bin/node" "/config/node-red/node_modules/.bin/node-gyp" "rebuild"
2022-12-11T19:54:31.869Z [err] npm ERR! gyp ERR! cwd /config/node-red/node_modules/@serialport/bindings
2022-12-11T19:54:31.869Z [err] npm
2022-12-11T19:54:31.869Z [err]  ERR! gyp ERR! node -v v16.17.1
2022-12-11T19:54:31.869Z [err] npm ERR! gyp ERR! node-gyp -v v3.8.0
2022-12-11T19:54:31.869Z [err] npm ERR! gyp ERR! not ok
2022-12-11T19:54:31.875Z [err] 
2022-12-11T19:54:31.876Z [err] npm ERR!
2022-12-11T19:54:31.876Z [err]  A complete log of this run can be found in:
2022-12-11T19:54:31.876Z [err] npm ERR!     /root/.npm/_logs/2022-12-11T19_54_08_687Z-debug-0.log
2022-12-11T19:54:31.923Z rc=1

Zahnweh, 2022-12-12:
Same here. Home Assistant OS 2022.12.1 with Node Red Add-On 13.5.3.

Zahnweh, 2022-12-12:
Duplicate of https://github.com/coolchip/node-red-contrib-smartmeter/issues/47