coolchip/node-red-contrib-smartmeter #49: lack of error handling causes node-red crash

ctr49, 2023-01-11:
I recently installed node-red-contrib-smartmeter on a system with:
Node-RED version: v3.0.2
Node.js  version: v18.12.1
Linux 5.15.74-1-pve x64 LE
node-red-contrib-smartmeter: v0.6.3

Unfortunately there seems to be an issue with both of my smart meters (Hager eHZ and Iskra MT681) and/or the serial connection and/or permissions or lxc device pass-through. I still have to figure out what the actual problem is, but whatever it is, it brings down the entire module and node-red as result (resulting in a restart loop until I disconnect the device):

error messages from log:
```
Jan 11 17:41:15 node-red node-red[582]: 11 Jan 17:41:15 - [red] Uncaught Exception:
Jan 11 17:41:15 node-red node-red[582]: 11 Jan 17:41:15 - [error] TypeError: smlFile.messages[msg].getMessageTag is not a function
Jan 11 17:41:15 node-red node-red[582]:     at SmlProtocol.handleMessage (/opt/nodered/node_modules/smartmeter-obis/lib/protocols/SmlProtocol.js:98:39)
Jan 11 17:41:15 node-red node-red[582]:     at SerialResponseTransport.processData (/opt/nodered/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:148:37)
Jan 11 17:41:15 node-red node-red[582]:     at SerialPort.<anonymous> (/opt/nodered/node_modules/smartmeter-obis/lib/transports/SerialResponseTransport.js:83:25)
Jan 11 17:41:15 node-red node-red[582]:     at SerialPort.emit (node:events:513:28)
Jan 11 17:41:15 node-red node-red[582]:     at SerialPort.emit (node:domain:489:12)
Jan 11 17:41:15 node-red node-red[582]:     at addChunk (node:internal/streams/readable:324:12)
Jan 11 17:41:15 node-red node-red[582]:     at readableAddChunk (node:internal/streams/readable:297:9)
Jan 11 17:41:15 node-red node-red[582]:     at SerialPort.Readable.push (node:internal/streams/readable:234:10)
Jan 11 17:41:15 node-red node-red[582]:     at /opt/nodered/node_modules/@serialport/stream/lib/index.js:385:12
```

At this point I'm not looking for help to troubleshoot the issue (I'm sure it's something I can easily find out), but want to report the issue that an error in processing a message / making a connection (this really only happens when a meter is connected on the serial port) may bring down the module, causing an availability issue.

Let me know if you need any additional information.

w4tsn, 2023-02-03:
I tried to understand where one would catch that exception, but I don't. Apparently this exception is thrown in the smartmeter-obis code, not the node-red node. After the node-red node calls the `init()` function of the smartmeter-obis module the serial port code is tied to the smartmeter-obis code as I understand it. So if something is emitted from the serial port it goes directly into the smartmeter-obis code and I don't know how to catch that from this project :thinking:

coolchip, 2023-02-08:
Thanks @ctr49 for publishing this issue for other users and thank you, @w4tsn, very much for trying to answer. 
It is right, that this seems to be a problem at the smartmeter-obis module. But I also don't know, why this is happening.
I also have no experiance in using lxc with serial devices.