coolchip/node-red-contrib-smartmeter #50: Do data from sml smartmeter to nodered

Reiseradler, 2023-02-03:
Hello,

I am running noderd on a raspberry pi 4 as docker container (as well as mosquitto, timescaledb and grafana). But I fail to get data from a smartmeter (sml) into nodered. The raspi is connected to an optical sensor via usb cable and I get data on the raspi (sudo cat /dev/ttyUSB0 | od -tx1).

I do not find any parameter configuration for the smartmeter node to get any data into nodered. Below you see the 2 nodes flow as well as the properties of the smartmeter node (connection details: 9600 Baud rate, 8N1 - should be fine since it is from the manual and already worked before).

Thank you very much in advance!

[image]
[image]
[image]

coolchip, 2023-02-08:
I've never done this but you have to give the docker container access to your serial device (/dev/ttyUSB0). Without this, the docker container isn't able to use it. I've found this with a google search:
https://www.losant.com/blog/how-to-access-serial-devices-in-docker

Reiseradler, 2023-02-12:
Thank you very much for your help! Due to your hint I am already one step further. 

To check the serial device connection with the docker container I have installed serialport in nodered. After some adjustments (I am not anymore sure which were the important ones) the serialport node in nodered could connect to /dev/ttyUSB0. Now, I get values - strang ones - from my serial device into nodered.

But for the smartmeter node I still get no values even though the parameters are the same than from the serialport node. Do you have any idea?
[image]

coolchip, 2023-02-18:
Did you completely disable or remove the serial node before using the smartmeter-node? Only one can access a serial port at a time.

Reiseradler, 2023-02-20:
Thank you so much!!!
Since deleting the test node "serialport", everything works perfect.

ToboCop81, 2023-02-20:
I have the exact same problem - at first I used the wrong serial port -  but I was able to fix it. Then I tested it using the serial node and encoded SML data. 
But when use the smartmeter node it only shows the following debug message after a view minutes:

~~~
"smartmeter-obis error: Error: No or too long answer from Serial Device after last request."
~~~

**And yes I did delete the serial node and its config node before I used the smartmeter node.**
I have an SGM-C4 Meter: 
https://www.efr.de/en/products/smart-energy-devices/electronic-electricity-meters/mme-en/sgm-c4/

ToboCop81, 2023-02-20:
Here are some scrrenshots of my configuration of the serial node and the payload which is returned in this case:

[image]

[image]

coolchip, 2023-02-20:
> Thank you so much!!! Since deleting the test node "serialport", everything works perfect.

That's fine. Enjoy it.

coolchip, 2023-02-20:
> Here are some scrrenshots of my configuration of the serial node and the payload which is returned in this case:
> 

What is your configuration of the smartmeter-node?

ToboCop81, 2023-02-20:
It is the same configuration which @Reiseradler uses:
[image]

coolchip, 2023-02-21:
The manual of your smartmeter says, that it is using the D0 protocol on the optical interface. You have selected "sml".
[image]

ToboCop81, 2023-02-22:
Ok that's interesting - because I found a hint to SML in this product manual:
[FR-SGM-C4-Produkthandbuch.pdf](https://www.mit-n.de/fileadmin/user_upload/Dateien/Messwesen/Messwesen_Strom/EFR-SGM-C4-Produkthandbuch.pdf#Page=32)

Can you provide a link to this manual you found please?
I will try the D0 protocol settings and let you know.

ToboCop81, 2023-02-22:
Hi - ich schreibe jetzt auch mal auf Deutsch weiter... Habe die Angabe in dem Datenblatt gefunden - da steht tatsächlich D0 Protokoll. - Im Produkthandbuch steht aber, dass die optische Schnittstelle mit SML arbeitet.. da gibt es wohl einen Widerspruch in der Doku...

Habe mal auf D0 umgestellt - das ist das Ergebnis:
[image]

ToboCop81, 2023-02-22:
Dieser Forums-Thread könnte evtl auch hilfreich sein (ich selbst steige da noch nicht ganz durch):
https://www.photovoltaikforum.com/thread/158054-libsml-error-unrecognized-sequence-beim-smartmeter-efr-sgm-c4/

Wäre für noch einen Tipp Dankbar :-)