coolchip/node-red-contrib-smartmeter #57: Not working with node-red v3.1.0

abergmann, 2023-09-28:
There is an issue with the `node-red-contrib-smartmeter` (v0.6.3) module in combination with `node-red` (v3.1.0).

Inside the "Palette" view the module can be installed, but has a warning displayed.
```
📦 node-red-contrib-smartmeter
🏷️ 0.6.3
   [object Object]
⚠️ [object Object]
```
I'm currently unsure how to extract more information out of `node-red` to identify the issue.

Any help would be appreciated...

coolchip, 2023-11-04:
I've tested it with a fresh Node-RED v3.1.0 installation and there everything looks right at the "Palette". 
[image]

NikAWing, 2024-04-27:
I have the same problem
NPM says: 
# npm audit report

request  *
Severity: moderate
Server-Side Request Forgery in Request - https://github.com/advisories/GHSA-p8p7-x288-28g6
Depends on vulnerable versions of tough-cookie
No fix available
node_modules/request
  smartmeter-obis  *
  Depends on vulnerable versions of request
  node_modules/smartmeter-obis
    node-red-contrib-smartmeter  *
    Depends on vulnerable versions of smartmeter-obis
    node_modules/node-red-contrib-smartmeter

tough-cookie  <4.1.3
Severity: moderate
tough-cookie Prototype Pollution vulnerability - https://github.com/advisories/GHSA-72xf-g2v4-qvf3
No fix available
node_modules/tough-cookie

4 moderate severity vulnerabilities

Some issues need review, and may require choosing
a different dependency.