coolchip/node-red-contrib-smartmeter #59: Stopped working after restart

greinet, 2023-11-28:
Hello,

I have been using this module for quite some time now and it was always working great, thanks for developing! However, today I had to restart the Raspberry Pi it is running on and it suddenly stopped working, I don't get any data anymore. 

I am using the following configuration:
[image]

I enabled debugging and this is the output, for me it seems like at least the data is coming in (shortened version):

```
-- Journal begins at Mon 2023-10-30 09:20:08 GMT, ends at Tue 2023-11-28 09:19:14 GMT. --
Nov 28 09:19:12 raspberrypi Node-RED[18172]: MATCH-RESULT SIGNON: "8.0*97(00086.9*kWh)\r\n1-0:1.8.0*98(00360.4*kWh)\r\n1-0:1.8.0*99(02770.7*kWh)......
Nov 28 09:19:12 raspberrypi Node-RED[18172]: NEW DATA
Nov 28 09:19:12 raspberrypi Node-RED[18172]: CURRENT PROCESS STEP 2 IN CHECKMESSAGE
Nov 28 09:19:12 raspberrypi Node-RED[18172]: MATCH-RESULT SIGNON: "8.0*97(00086.9*kWh)\r\n1-0:1.8.0*98(00360.4*kWh)\r\n1-0:1.8.0*99(02770.7*kWh)......
Nov 28 09:19:12 raspberrypi Node-RED[18172]: NEW DATA
Nov 28 09:19:12 raspberrypi Node-RED[18172]: CURRENT PROCESS STEP 2 IN CHECKMESSAGE
Nov 28 09:19:12 raspberrypi Node-RED[18172]: MATCH-RESULT SIGNON: "8.0*97(00086.9*kWh)\r\n1-0:1.8.0*98(00360.4*kWh)\r\n1-0:1.8.0*99(02770.7*kWh).........
```

Full log output: [logs.txt](https://github.com/coolchip/node-red-contrib-smartmeter/files/13486060/logs.txt)

To my knowledge nothing happened to the system other than the simple restart. Any idea what could be causing this issue or how I can debug it further?

Also, I'm not sure if this is a problem with the NodeRED part or the library itself (or on my side), please tell my if I should open an issue at the core library.

Thanks in advance,
Andy

greinet, 2023-11-28:
After lots of trying around, finally switching to a different Raspberry Pi got my flow working again, no idea what the original issue was but i could not reproduce it.

coolchip, 2024-02-25:
Strange. But it's fine that you was able to fix it.