coolchip/node-red-contrib-smartmeter #60: Installation not working with NodeJS 20.

Horscht1985, 2023-12-19:
I can't install the Plugin with NodeJS20:

`NRTest:~# node -v
v20.10.0
`

`NRTest:~# npm install node-red-contrib-smartmeter
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code 1
npm ERR! path /root/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! make: Entering directory '/root/node_modules/@serialport/bindings/build'
npm ERR!   CXX(target) Release/obj.target/bindings/src/serialport.o
npm ERR!   CXX(target) Release/obj.target/bindings/src/serialport_unix.o
npm ERR! make: Leaving directory '/root/node_modules/@serialport/bindings/build'
npm ERR! prebuild-install warn install No prebuilt binaries found (target=20.10.0 runtime=node arch=x64 libc=musl platform=linux)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@10.0.1
npm ERR! gyp info using node@20.10.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.11.6 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args 'binding.gyp',
npm ERR! gyp info spawn args '-f',
npm ERR! gyp info spawn args 'make',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/root/node_modules/@serialport/bindings/build/config.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/root/.cache/node-gyp/20.10.0/include/node/common.gypi',
npm ERR! gyp info spawn args '-Dlibrary=shared_library',
npm ERR! gyp info spawn args '-Dvisibility=default',
npm ERR! gyp info spawn args '-Dnode_root_dir=/root/.cache/node-gyp/20.10.0',
npm ERR! gyp info spawn args '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args '-Dnode_lib_file=/root/.cache/node-gyp/20.10.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args '-Dmodule_root_dir=/root/node_modules/@serialport/bindings',
npm ERR! gyp info spawn args '-Dnode_engine=v8',
npm ERR! gyp info spawn args '--depth=.',
npm ERR! gyp info spawn args '--no-parallel',
npm ERR! gyp info spawn args '--generator-output',
npm ERR! gyp info spawn args 'build',
npm ERR! gyp info spawn args '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/./serialport.h:13,
npm ERR!                  from ../src/serialport.cpp:1:
npm ERR! ../../../nan/nan.h: In function 'bool Nan::IdleNotification(int)':
npm ERR! ../../../nan/nan.h:688:63: warning: 'bool v8::Isolate::IdleNotificationDeadline(double)' is deprecated: Use MemoryPressureNotification() to influence the GC schedule. [-Wdeprecated-declarations]
npm ERR!   688 |     return v8::Isolate::GetCurrent()->IdleNotificationDeadline(
npm ERR!       |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
npm ERR!   689 |         idle_time_in_ms * 0.001);
npm ERR!       |         ~~~~~~~~~~~~~~~~~~~~~~~~                               
npm ERR! In file included from /root/.cache/node-gyp/20.10.0/include/node/v8-initialization.h:13,
npm ERR!                  from /root/.cache/node-gyp/20.10.0/include/node/v8.h:34,
npm ERR!                  from /root/.cache/node-gyp/20.10.0/include/node/node.h:73,
npm ERR!                  from ../../../nan/nan.h:62:
npm ERR! /root/.cache/node-gyp/20.10.0/include/node/v8-isolate.h:1293:8: note: declared here
npm ERR!  1293 |   bool IdleNotificationDeadline(double deadline_in_seconds);
npm ERR!       |        ^~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! ../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
npm ERR! ../src/serialport.cpp:78:51: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
npm ERR!    78 |   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
npm ERR!       |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! ../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)':
[... 42 lines cut ...]
npm ERR! /root/.cache/node-gyp/20.10.0/include/node/node.h:1172:7: warning: cast between incompatible function types from 'void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)' {aka 'void (*)(v8::Local<v8::Object>)'} to 'node::addon_register_func' {aka 'void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)'} [-Wcast-function-type]
npm ERR!  1172 |       (node::addon_register_func) (regfunc),                          \
npm ERR!       |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! /root/.cache/node-gyp/20.10.0/include/node/node.h:1206:3: note: in expansion of macro 'NODE_MODULE_X'
npm ERR!  1206 |   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
npm ERR!       |   ^~~~~~~~~~~~~
npm ERR! ../src/serialport.cpp:486:1: note: in expansion of macro 'NODE_MODULE'
npm ERR!   486 | NODE_MODULE(serialport, init);
npm ERR!       | ^~~~~~~~~~~
npm ERR! ../src/serialport_unix.cpp:27:10: fatal error: linux/serial.h: No such file or directory
npm ERR!    27 | #include <linux/serial.h>
npm ERR!       |          ^~~~~~~~~~~~~~~~
npm ERR! compilation terminated.
npm ERR! make: *** [bindings.target.mk:113: Release/obj.target/bindings/src/serialport_unix.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:209:23)
npm ERR! gyp ERR! System Linux 6.5.11-7-pve
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /root/node_modules/@serialport/bindings
npm ERR! gyp ERR! node -v v20.10.0
npm ERR! gyp ERR! node-gyp -v v10.0.1
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in: /root/.npm/_logs/2023-12-19T08_05_09_007Z-debug-0.log`

coolchip, 2024-02-25:
I've started the node:20-bookworm container and I'm able to install this node.
Do you use an x86_64 or amd64 arch cpu?

coolchip, 2024-02-25:
Okay... with node:20-alpine and `apk add build-base` and `apk add python3` I got an error which is looking similar to yours.

That's because of the missing linux headers. Add them with `apk add linux-headers`.

Generally I think running Node-RED on the small apline linux isn't a good idea. Better use debian, ubuntu ora a similar dist with a richer set of installed packages.

Horscht1985, 2024-02-27:
You're correct. I added them with `ask add linux-headers` and reinstalled the plugin => worked.

closed 2024-02-27 by Horscht1985