coolchip/node-red-contrib-smartmeter #61: Errors are not catched by catch node

Piranha1511, 2024-01-05:
My Smartmeter throws sporadicly a message which do not match the SML protocol. I want to catch this error with the error catch node in node red. This is not possible, since the error handling of the smartmeter node seems to be not correct implemented.

This is filling my debug panel in seconds. 

[image]

Piranha1511, 2024-01-21:
It also fills the syslog and usrlog of the system!

coolchip, 2024-02-25:
I understand your problem but I don't see any issue in the error handling implementation.
When you get this errors? Perhaps your sensor has a problem.

kopierschnitte, 2024-03-16:
I also see this very frequently on my Victron Cerbo GX, querying a Tibber Pulse (locally). Catch node doesn't seem to fire.
Maybe it's caused by piping the HTTP output to the smartmeter-node using a local file?

LoQue90, 2024-07-08:
i do have the same issue, also node red on victron multiplus gx. 
At some point this slows down node red massivly and adding delay to the messurements.
@kopierschnitte any luck on that?

kopierschnitte, 2024-07-08:
Not really. For now I'm just checking the statusCode of the HTTP request just before writing the data to the temp file (and then feeding this into the smartmeter-node). This way I can be sure that only valid responses are piped to the node:
[image]

But nevertheless, I still get debug errors every 20-30s and that is "resetting" my ESS logic. If there would be a way to properly catch these errors, I could use the measurements of a different device in the meantime.

kopierschnitte, 2024-10-19:
After testing, it turns out that the only possibility to avoid spamming the debug log was to disable to error reporting on code level.
Therefore, I've directly patched/commented the following line in LocalFileTransport.js:
`// self.protocol.callUserCallback(new Error('Received Message do not match protocol: ' + data), null);`

It's a dirty hack but it works...