coolchip/node-red-contrib-smartmeter #62: Invalid protocol

dibadodo, 2024-02-03:
I am trying to query a Tibber pulse. I have created the http host, entered the port, selected sml as the protocol and http request as the transport 
But I don't get any data. According to the network trace, there is also no request from NodeRed to the host.
In the console of Nodered I see the following message

ERROR ON HTTP REQUEST: : Error: Invalid protocol:

coolchip, 2024-02-25:
Which protocol and transport did you select in the node configuration?

dibadodo, 2024-03-01:
both was http 
But the other problem is, that a authentication during the http access is not possible with smartmeter
the best would be, when you can take the data from a previous node, like each other on node red . I can then make the pull with a specific http request an push the answer to the smartmeter node. I think this request was coming from other user to.

LoQue90, 2024-07-08:
i´m trying to do the same. Pulling the data from the tibber brigde and send it via mqtt to victron.
right now i´m pulling with a http node via get and store it into a file. smlparser is reading the file.
Any way to do this without writing it into a file?

dibadodo, 2024-07-20:
> i´m trying to do the same. Pulling the data from the tibber brigde and send it via mqtt to victron. right now i´m pulling with a http node via get and store it into a file. smlparser is reading the file. Any way to do this without writing it into a file?

You can do that with TCP connection - use TCP connector on same Port as you say it inside smart meter configuration. use 127.0.0.1 as ip address