coolchip/node-red-contrib-smartmeter #63: Could not install node-red-contrib-smartmeter in Docker

1einmal1, 2024-02-07:
Hi,
after updating Nodered from V2.2.3-12 to V3.1.3 in Docker (19.03.15) I could not more install the node-red-contrib-smartmeter V0.6.3
Docker is running on amd64 engine
Here the logfile:
> node-red-docker@3.1.3 start
> node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS --userDir /data
1 Feb 17:19:16 - [info] 
Welcome to Node-RED
===================
1 Feb 17:19:16 - [info] Node-RED version: v3.1.3
1 Feb 17:19:16 - [info] Node.js  version: v16.20.2
1 Feb 17:19:16 - [info] Linux 4.9.0-19-amd64 x64 LE
1 Feb 17:19:19 - [info] Loading palette nodes
1 Feb 17:19:22 - [info] Dashboard version 3.6.2 started at /ui
1 Feb 17:19:23 - [info] Settings file  : /data/settings.js
1 Feb 17:19:23 - [info] Context store  : 'default' [module=memory]
1 Feb 17:19:23 - [info] User directory : /data
1 Feb 17:19:23 - [warn] Projects disabled : editorTheme.projects.enabled=false
1 Feb 17:19:23 - [info] Flows file     : /data/flows.json
1 Feb 17:19:23 - [warn] 
1 Feb 17:19:23 - [warn] Encrypted credentials not found
1 Feb 17:19:23 - [info] Server now running at http://127.0.0.1:1880/
1 Feb 17:19:23 - [info] Starting flows
1 Feb 17:19:23 - [info] Started flows
1 Feb 17:20:17 - [info] Installing module: node-red-contrib-smartmeter, version: 0.6.3
-----------------------------------------------------------
2024-02-01T17:20:18.242Z Installieren : node-red-contrib-smartmeter 0.6.3

2024-02-01T17:20:17.365Z npm install --no-audit --no-update-notifier --no-fund --save --save-prefix=~ --production --engine-strict node-red-contrib-smartmeter@0.6.3
2024-02-01T17:20:19.698Z [err] npm
2024-02-01T17:20:19.699Z [err]  WARN
2024-02-01T17:20:19.700Z [err]  config production Use `--omit=dev` instead.
2024-02-01T17:20:33.979Z [err] npm
2024-02-01T17:20:33.980Z [err]  WARN deprecated har-validator@5.1.5: this library is no longer supported
2024-02-01T17:20:34.302Z [err] npm
2024-02-01T17:20:34.303Z [err]  WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
2024-02-01T17:20:34.387Z [err] npm 
2024-02-01T17:20:34.388Z [err] WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
2024-02-01T17:20:43.633Z [err] npm
2024-02-01T17:20:43.634Z [err]  ERR! code 1
2024-02-01T17:20:43.634Z [err] npm ERR! path /data/node_modules/@serialport/bindings
2024-02-01T17:20:43.645Z [err] npm ERR!
2024-02-01T17:20:43.646Z [err]  command failed
2024-02-01T17:20:43.646Z [err] npm ERR! command sh -c -- prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
2024-02-01T17:20:43.646Z [err] npm ERR!
2024-02-01T17:20:43.646Z [err]  make: Entering directory '/data/node_modules/@serialport/bindings/build'
2024-02-01T17:20:43.646Z [err] npm ERR! make: Leaving directory '/data/node_modules/@serialport/bindings/build'
2024-02-01T17:20:43.647Z [err] npm
2024-02-01T17:20:43.648Z [err]  
2024-02-01T17:20:43.648Z [err] ERR! prebuild-install warn install No prebuilt binaries found (target=16.20.2 runtime=node arch=x64 libc=musl platform=linux)
2024-02-01T17:20:43.648Z [err] npm ERR!
2024-02-01T17:20:43.648Z [err]  gyp info it worked if it ends with ok
2024-02-01T17:20:43.648Z [err] npm ERR! gyp info using node-gyp@9.1.0
2024-02-01T17:20:43.648Z [err] npm ERR!
2024-02-01T17:20:43.649Z [err]  gyp info using node@16.20.2 | linux | x64
2024-02-01T17:20:43.649Z [err] npm ERR! gyp info find Python using Python version 3.11.6 found at "/usr/bin/python3"
2024-02-01T17:20:43.649Z [err] npm 
2024-02-01T17:20:43.649Z [err] ERR! gyp http GET https://unofficial-builds.nodejs.org/download/release/v16.20.2/node-v16.20.2-headers.tar.gz
2024-02-01T17:20:43.649Z [err] npm ERR! gyp http 200 https://unofficial-builds.nodejs.org/download/release/v16.20.2/node-v16.20.2-headers.tar.gz
2024-02-01T17:20:43.649Z [err] npm
2024-02-01T17:20:43.650Z [err]  ERR! gyp http GET https://unofficial-builds.nodejs.org/download/release/v16.20.2/SHASUMS256.txt
2024-02-01T17:20:43.650Z [err] npm ERR! gyp http 200 https://unofficial-builds.nodejs.org/download/release/v16.20.2/SHASUMS256.txt
2024-02-01T17:20:43.650Z [err] npm ERR! gyp info spawn /usr/bin/python3
2024-02-01T17:20:43.650Z [err] npm ERR!
2024-02-01T17:20:43.650Z [err]  gyp info spawn args [
2024-02-01T17:20:43.650Z [err] npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
2024-02-01T17:20:43.650Z [err] npm ERR!
2024-02-01T17:20:43.651Z [err]  gyp info spawn args   'binding.gyp',
2024-02-01T17:20:43.651Z [err] npm ERR! gyp info spawn args   '-f',
2024-02-01T17:20:43.651Z [err] npm
2024-02-01T17:20:43.652Z [err]  ERR! gyp info spawn args   'make',
2024-02-01T17:20:43.652Z [err] npm ERR! gyp info spawn args   '-I',
2024-02-01T17:20:43.652Z [err] npm ERR! gyp info spawn args   '/data/node_modules/@serialport/bindings/build/config.gypi',
2024-02-01T17:20:43.652Z [err] npm ERR! gyp info spawn args   '-I',
2024-02-01T17:20:43.652Z [err] npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
2024-02-01T17:20:43.652Z [err] npm ERR! gyp info spawn args   '-I',
2024-02-01T17:20:43.652Z [err] npm
2024-02-01T17:20:43.653Z [err]  ERR! gyp info spawn args   '/usr/src/node-red/.cache/node-gyp/16.20.2/include/node/common.gypi',
2024-02-01T17:20:43.653Z [err] npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
2024-02-01T17:20:43.653Z [err] npm ERR! gyp info spawn args   '-Dvisibility=default',
2024-02-01T17:20:43.653Z [err] npm
2024-02-01T17:20:43.653Z [err]  ERR! gyp info spawn args   '-Dnode_root_dir=/usr/src/node-red/.cache/node-gyp/16.20.2',
2024-02-01T17:20:43.653Z [err] npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
2024-02-01T17:20:43.653Z [err] npm ERR! gyp info spawn args   '-Dnode_lib_file=/usr/src/node-red/.cache/node-gyp/16.20.2/<(target_arch)/node.lib',
2024-02-01T17:20:43.653Z [err] npm 
2024-02-01T17:20:43.654Z [err] ERR! gyp info spawn args   '-Dmodule_root_dir=/data/node_modules/@serialport/bindings',
2024-02-01T17:20:43.654Z [err] npm ERR! gyp info spawn args   '-Dnode_engine=v8',
2024-02-01T17:20:43.654Z [err] npm
2024-02-01T17:20:43.654Z [err]  ERR! gyp info spawn args   '--depth=.',
2024-02-01T17:20:43.654Z [err] npm ERR! gyp info spawn args   '--no-parallel',
2024-02-01T17:20:43.654Z [err] npm
2024-02-01T17:20:43.655Z [err]  ERR! gyp info spawn args   '--generator-output',
2024-02-01T17:20:43.655Z [err] npm ERR! gyp info spawn args   'build',
2024-02-01T17:20:43.655Z [err] npm ERR! gyp info spawn args   '-Goutput_dir=.'
2024-02-01T17:20:43.655Z [err] npm 
2024-02-01T17:20:43.655Z [err] ERR! gyp info spawn args ]
2024-02-01T17:20:43.655Z [err] npm ERR! gyp info spawn make
2024-02-01T17:20:43.655Z [err] npm ERR!
2024-02-01T17:20:43.656Z [err]  gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
2024-02-01T17:20:43.656Z [err] npm ERR! make: printf: Operation not permitted
2024-02-01T17:20:43.656Z [err] npm
2024-02-01T17:20:43.656Z [err]  ERR! make: *** [bindings.target.mk:116: Release/obj.target/bindings/src/serialport.o] Error 127
2024-02-01T17:20:43.656Z [err] npm ERR! gyp ERR! build error 
2024-02-01T17:20:43.656Z [err] npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
2024-02-01T17:20:43.656Z [err] npm 
2024-02-01T17:20:43.657Z [err] ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:201:23)
2024-02-01T17:20:43.657Z [err] npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
2024-02-01T17:20:43.657Z [err] npm
2024-02-01T17:20:43.657Z [err]  ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:293:12)
2024-02-01T17:20:43.657Z [err] npm ERR!
2024-02-01T17:20:43.657Z [err]  gyp ERR! System Linux 4.9.0-19-amd64
2024-02-01T17:20:43.657Z [err] npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
2024-02-01T17:20:43.657Z [err] npm 
2024-02-01T17:20:43.658Z [err] ERR! gyp ERR! cwd /data/node_modules/@serialport/bindings
2024-02-01T17:20:43.658Z [err] npm ERR! gyp ERR! node -v v16.20.2
2024-02-01T17:20:43.658Z [err] npm ERR! gyp ERR! node-gyp -v v9.1.0
2024-02-01T17:20:43.658Z [err] npm 
2024-02-01T17:20:43.658Z [err] ERR! gyp ERR! not ok
2024-02-01T17:20:43.669Z [err] 
2024-02-01T17:20:43.670Z [err] npm 
2024-02-01T17:20:43.671Z [err] ERR! A complete log of this run can be found in:
2024-02-01T17:20:43.671Z [err] npm ERR!     /data/.npm/_logs/2024-02-01T17_20_19_504Z-debug-0.log
2024-02-01T17:20:43.717Z rc=1

coolchip, 2024-02-25:
I use the docker image nodered/node-red at version 3.1.3 on my x86_64 CPU. I got no error when installing this node. 

Usually this errors are a result of missing build tools. On Debian/Ubuntu you need the `build-essential` package. On Alpine it should be `build-base`.

1einmal1, 2024-02-26:
Yes, on my test system with the newest updates I run the the node without problems.  
But on the production host system I run a very old Debian (Linux 4.9.0-19-amd64 x64 LE) on a Rpi4 and I have no chance to update the host and docker due to missing access to the system. 
To be independent from the host system the missing  build-essential package should be integrated to the Node Red module, if possible. 
Thanks

coolchip, 2024-02-26:
Unfortunatelly this isn't possible. The `build-essential` is a Debian package. Not an NPM package.

coolchip, 2024-02-27:
And every NPN package, that contains c++ code, needs this. That is not a problem specific to this node. 😉