coolchip/node-red-contrib-smartmeter #65: Compatibility with Node-RED 4.0

MPueschel, 2024-06-20:
Hi,

the Node does not seem to be compatible with Node-RED 4.0. Is there a simple solution for this? Thank you.

ThomasCr, 2024-06-27:
I have now the same problem... :(

jgantenberg, 2024-06-28:
There seems to be an issue with the upgrade from node-red 3.1 to 4.0. I run node-red with smartmeter module in a docker container and the upgrade to node-red 4.0 brought the discussed error. But now I set up a new blank container, installed all needed modules via palette and imported my flows from the old container. As usual I had to fill credentials manually but after that everything works fine.

MatzeG, 2024-06-28:
thx jgantenberg

also first uninstalling only the smartmeter component, updating to nodered 4.0.x and reinstalling the smartmeter works for me in the docker environment

MPueschel, 2024-07-01:
thx jgantenberg and MatzeG

MatzeG's approach worked perfectly for me.

I previously exported the smartmeter-node config and imported it again after updating and reinstalling.